FreeCAD/FreeCAD #9859: [Problem] Sketcher toolbar re-enabled when switching tab with "Remember WB by tab" active

0penBrain, 2023-07-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.2 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33482 (Git)
Build type: Debug
Branch: master
Hash: 3d0fe64c6652eaff4ff8fc105bd7ab53f021e82d
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: French/France (fr_FR)
```

### Subproject(s) affected?

Sketcher

### Problem description

Go to Preferences and enable Workbenches -> "Remember active workbench by tab"

Create a new document (1)
Create another document (2)

In document 2, switch to Part WB

Then in document 1, switch to Sketcher WB and create a new Sketch (stay in Edit mode).
Switch to document 2 view/tab.
Switch back to document 1 view/tab -> :warning: `[Sketcher]` toolbar has been enabled by the switch while it should be kept hidden until Edit mode is left.

### Anything else?

This report will be true in its behavior only after #9758 merged.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2023-07-01:
@PaddleStroke please consider fixing this regression. Thx

luzpaz, 2023-10-29:
ping @PaddleStroke

0penBrain, 2025-02-22:
Still exists in
OS: Ubuntu 22.04.4 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40326 (Git)
Build type: Debug
Branch: main
Hash: f74efa319c20e915e7a7122b8ee7d4499d6bffe2
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Logical/physical DPI: 96.1263/141.585

FEA-eng, 2025-06-08:
Still relevant

closed 2025-09-30 by chennes