FreeCAD/FreeCAD #9860: [Problem]  Latest (0.21.33458) Mac-X86 crashes on file save

edwilliams16, 2023-07-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 13.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * DynamicData 2.46.0
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure-main
  * GDML 1.8.0
  * offline-documentation
  * Manipulator 1.5.0
  * fasteners 0.4.56
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.12
[/code]
```

### Subproject(s) affected?

None

### Problem description

Open any file and attempt to save it.  FreeCAD crashes.  
See [https://forum.freecad.org/viewtopic.php?t=79384](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-07-02:
can you run with logging enabled and upload the log?

edwilliams16, 2023-07-02:
[FreeCADx86log.txt](https://github.com/FreeCAD/FreeCAD/files/11928544/FreeCADx86log.txt)

Here's the log.  I opened a file and tried to save it - but I don't see it being logged.

edwilliams16, 2023-07-02:
Here's Apple Crash log.  I rebooted before I did this in case the OS had a problem.  Didn't seem to be the case.
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [1089]
Path:                  /Applications/FreeCADx86.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        Exited process [1082]
User ID:               501

Date/Time:             2023-07-01 16:41:02.5418 -1000
OS Version:            macOS 13.2.1 (22D68)
Report Version:        12
Anonymous UUID:        D6C571A8-2D94-0C29-A3D0-5B218F76898B

Time Awake Since Boot: 220 seconds

System Integrity Protection: enabled

Notes:
dyld_process_snapshot_create_for_process failed with 5

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Codes:       0x0000000000000001, 0x0000000000000028

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1089]

VM Region Info: 0x28 is not in any region.  Bytes before following region: 140722839306200
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      mapped file              7ffc96d94000-7ffca1860000 [170.8M] r-x/r-x SM=COW  ...t_id=381ed3ab

Error Formulating Crash Report:
dyld_process_snapshot_create_for_process failed with 5

Kernel Triage:
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libc++.1.dylib                	    0x7ff80c1ff058 std::__1::istreambuf_iterator<char, std::__1::char_traits<char> > std::__1::num_get<char, std::__1::istreambuf_iterator<char, std::__1::char_traits<char> > >::__do_get_signed<long>(std::__1::istreambuf_iterator<char, std::__1::char_traits<char> >, std::__1::istreambuf_iterator<char, std::__1::char_traits<char> >, std::__1::ios_base&, unsigned int&, long&) const + 54
1   libc++.1.0.dylib              	       0x111c4266a std::__1::basic_ostream<char, std::__1::char_traits<char> >::operator<<(int) + 282
2   libTKBRep.7.6.3.dylib         	       0x19f1a6845 TopTools_ShapeSet::Write(TopoDS_Shape const&, std::__1::basic_ostream<char, std::__1::char_traits<char> >&) const + 309
3   Part.so                       	       0x12ab76f42 Part::TopoShape::exportBrep(std::__1::basic_ostream<char, std::__1::char_traits<char> >&) const + 130
4   Part.so                       	       0x12a96d62c Part::PropertyPartShape::SaveDocFile(Base::Writer&) const + 396
5   libFreeCADBase.dylib          	       0x10d8ba858 Base::ZipWriter::writeFiles() + 136
6   libFreeCADApp.dylib           	       0x11017dc5c App::Document::saveToFile(char const*) const + 1580
7   libFreeCADApp.dylib           	       0x11017d376 App::Document::save() + 1062
8   libFreeCADApp.dylib           	       0x11026e4c8 App::DocumentPy::save(_object*) + 56
9   libFreeCADApp.dylib           	       0x110268e1a App::DocumentPy::staticCallback_save(_object*, _object*) + 90
10  libpython3.10.dylib           	       0x10cfb338a cfunction_call + 90
11  libpython3.10.dylib           	       0x10d07c46f _PyEval_EvalFrameDefault + 73471
12  libpython3.10.dylib           	       0x10d06853d _PyEval_Vector + 541
13  libpython3.10.dylib           	       0x10d0eecb0 run_mod + 240
14  libpython3.10.dylib           	       0x10d0f0510 PyRun_StringFlags + 128
15  libFreeCADBase.dylib          	       0x10d8302fe Base::InterpreterSingleton::runString(char const*) + 94
16  libFreeCADGui.dylib           	       0x10e085f41 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 145
17  libFreeCADGui.dylib           	       0x10e085d7c Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) + 188
[... 909 lines cut ...]
      "deploymentId" : 400000012
    }
  ]
},
  "reportNotes" : [
  "dyld_process_snapshot_create_for_process failed with 5"
]
}

Model: MacBookPro18,1, BootROM 8419.80.7, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Nov 30 2022 02:17:16 version 20.10.965.13.8.7.131 FWID 01-1251c18d
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: SanDisk 3.2Gen1
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: ScanSnap iX500
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

edwilliams16, 2023-07-02:
It runs OK on my 13 year old Mac:
```
OS: macOS High Sierra (10.13)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * CurvedShapes 1.0.4
  * Curves 0.6.9
  * DynamicData 2.46.0
  * fasteners 0.4.56
  * fcgear 1.0.0
  * GDML 1.8.0
  * Help 1.0.3
  * lattice2 1.0.0
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * offline-documentation 1.0.0-alpha
  * Pyramids-and-Polyhedrons
  * QuickMeasure-main
  * toSketch 1.0.1
  * Trails 2022.1.0
```

adrianinsaval, 2023-07-02:
yeah it's compiled against mac 10.13 but it's supposed to work nonetheless on later versions, not sure what's the problem. Was there an OS update?

edwilliams16, 2023-07-02:
No updates between the last working FreeCAD version and this one (two weeks ago?) AFAIK.  I updated to Ventura 13.4.1 this AM to no effect.

turboooooo, 2023-07-02:
I have the same error on recent weekly builds (both arm + x86 mac builds) on mac os ventura 13.4.1. It will always crash on save except on empty files. As soon as it contains a sketch it will crash.

This is the the output in terminal
1. when I click on "create sketch":
FALLBACK (log once): Fallback to SW vertex for line stipple
FALLBACK (log once): Fallback to SW vertex processing, m_disable_code: 2000
FALLBACK (log once): Fallback to SW vertex processing in drawCore, m_disable_code: 2000

2. when I click on save and a save dialog pops up:
2023-07-03 00:27:49.050 freecad[35697:3701916] +[CATransaction synchronize] called within transaction
2023-07-03 00:27:49.098 freecad[35697:3701916] +[CATransaction synchronize] called within transaction
2023-07-03 00:27:49.100 freecad[35697:3701916] Warning: Window move completed without beginning
2023-07-03 00:27:54.079 freecad[35697:3701916] +[CATransaction synchronize] called within transaction

3. once I click save in the dialog it crashes immediatly with the follwing line:
/Applications/FreeCAD.app/Contents/MacOS/FreeCAD: line 19: 35527 Segmentation fault: 11 "${PREFIX}/bin/freecad" $@

The log does not show anything interesting and nothing on crash. I hope this helps? If there's an archive of weekly builds I could test and report since when this crash started to occur.

JohnOCFII, 2023-07-03:
I can confirm that saving files with latest Mac arm version also crashes for me.  For comparison, Mac arm version 33447 works fine.  I notice that `Vtk` and `OCC` versions differ between the two builds.

Here's the details first from the working version.

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33447 (Git)
Build type: Release
Branch: master
Hash: 090ae4571a589171dccb5dcabaf9d71e453ab2f3
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.7.1
Locale: C/Default (C)
```

And the version that crashes:

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
```

I get as far as entering a name into the save dialog box, press RETURN, and BOOM:

Here's the start of the crash report:

```
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Codes:       0x0000000000000001, 0x0000000000000028

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [66028]

VM Region Info: 0x28 is not in any region.  Bytes before following region: 105553518919640
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libc++.1.dylib                	       0x1a8eac7c4 std::__1::istreambuf_iterator<char, std::__1::char_traits<char>> std::__1::num_get<char, std::__1::istreambuf_iterator<char, std::__1::char_traits<char>>>::__do_get_signed<long>(std::__1::istreambuf_iterator<char, std::__1::char_traits<char>>, std::__1::istreambuf_iterator<char, std::__1::char_traits<char>>, std::__1::ios_base&, unsigned int&, long&) const + 72
1   libc++.1.0.dylib              	       0x106d52f9c std::__1::basic_ostream<char, std::__1::char_traits<char>>::operator<<(int) + 300
2   Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
3   Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
4   Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
5   Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
6   Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
7   Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
8   Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
9   Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
10  Sketcher.so                   	       0x1561cfd30 Sketcher::PropertyConstraintList::Save(Base::Writer&) const + 128
```

tonealone, 2023-07-03:
Just adding to this - from my forum post:
The latest weekly build crashes when saving a file.

FreeCAD info:
[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * fasteners 0.4.56
  * NikraDAP 2.0.0
  * Curves 0.6.12
  * Assembly4 0.50.2
[/code]

Seems the program doesn't crash unless there's a sketch in the file…?
If I create a file, then save - no crash
I then add a body, then save - no crash
I then add a sketch, then save - crash

I tried disabling:
"Save thumbnail into project file when saving document" 
and
"Add the program logo to the generated thumbnail"
but it didn't make a difference, it crashed whether selected or not…

Attached are the crash log files.

[freecad-2023-06-29-210548.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911508/freecad-2023-06-29-210548.txt)
[freecad-2023-06-29-210331.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911509/freecad-2023-06-29-210331.txt)
[freecad-2023-06-29-204533.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911510/freecad-2023-06-29-204533.txt)
[freecad-2023-06-29-202712.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911511/freecad-2023-06-29-202712.txt)
[freecad-2023-06-29-191438.txt](https://github.com/FreeCAD/FreeCAD-Bundle/files/11911512/freecad-2023-06-29-191438.txt)

adrianinsaval, 2023-07-03:
this might be a problem in how the bundles are made, however I don't think I'll be able to work on this until around friday.

JohnOCFII, 2023-07-03:
Tomorrow afternoon (US Central time) I'll try to do a Conda build locally (using the Oursland Build process https://github.com/oursland/FreeCAD-Build-Notes) and see if that differs in behavior.

JohnOCFII, 2023-07-04:
> Tomorrow afternoon (US Central time) I'll try to do a Conda build locally (using the Oursland Build process https://github.com/oursland/FreeCAD-Build-Notes) and see if that differs in behavior.

Unfortunately Oursland's build process is failing at this step:
```
(freecad)osh@OshMBPro FreeCAD % conda devenv -f conda/environment.devenv.yml
conda env update --file conda/environment.yml --prune

EnvironmentSectionNotValid: The following section on '/Users/osh/Documents/Dev/FreeCAD/conda/environment.yml' is invalid and will be ignored:
 - environment
```

Which shortly there after leads to:
```
                                                                                                                                   
Installing pip dependencies: - Ran pip subprocess with arguments:                                                                                       
['/Users/osh/Documents/Dev/FreeCAD/.conda/freecad/bin/python', '-m', 'pip', 'install', '-U', '-r', '/Users/osh/Documents/Dev/FreeCAD/conda/condaenv.f4yb4re0.requirements.txt', '--exists-action=b']                                                                                                            
Pip subprocess output:                                                                                                                                  
                                                                                                                                                        
Pip subprocess error:                                                                                                                                   
                                                                                                                                                        
failed                                                                                                                                                  
                                                                                                                                                        
CondaEnvException: Pip failed               
```

So I don't know if this is related to recent packaging changes/attempts or not.

alidaf, 2023-07-04:
This happens with the current weekly build 02/07/23 for me when saving to a SMB network drive.

Nailig, 2023-07-09:
same here, version 0.21.0-33492 on macOS 13.4.1 ARM 
I can save an empty project, but as soon as I add a sketch, a save crashes freecad.

I thought I would start learning free cad with a recent version..... better switch to a stable version :)

JohnOCFII, 2023-07-11:
I just tested the fresh weekly build on my M1/arm Mac, and was not able to replicate the problem. So for me it seems fixed.

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33516 (Git)
Build type: Release
Branch: master
Hash: c09ef943f76a9693af6a045220070f685847e458
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

adrianinsaval, 2023-07-11:
Good, this might have been just a discrepancy between the libs used during build and the libs that get packaged. X86_64 version might still have the problem as I'm having some trouble getting it to compile with the space mouse driver now

edwilliams16, 2023-07-11:
Mac Arm 33516 version no longer crashes on Save - still crashes using the transform tool.

turboooooo, 2023-07-11:
How do you get the Mac Arm 33516 running? For me it says "Freecad was blocked from opening because it is not from an identified developer.”? Disabling gatekeeper does not help as it should in this case. When I try to start the app from finder is says the package might be corrupted.

JohnOCFII, 2023-07-11:
First off - what OS are you running?   If running a Beta for Sonoma, I’m not sure if the procedure differs.   I’m running Ventura.

I follow these instructions (basically, right click and select Open) and it works for me.

edwilliams16, 2023-07-12:
Ive found to open FreeCAD in 13.4.2 i have to ctl-click Open, cancel, ctl-click open a second time, then open.  Once used to be enough.

adrianinsaval, 2023-07-13:
Can somebody please test the latest x86_64 dmg?

turboooooo, 2023-07-13:
Sure: FreeCAD_weekly-builds-33576-2023-07-13-conda-macOS-x86_64-py310.dmg works fine as well again.

adrianinsaval, 2023-07-13:
Excellent! Thanks for testing

closed 2023-07-13 by adrianinsaval

JohnOCFII, 2023-07-13:
> Ive found to open FreeCAD in 13.4.2 i have to ctl-click Open, cancel, ctl-click open a second time, then open. Once used to be enough.

Yes - I just realized I need to run through this step twice as well -- first time the option to `Open` is not shown.