FreeCAD/FreeCAD #9861: [Problem]  Clicking in field causes negative fractional value to increase

smrruby, 2023-07-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Whenever a value is set using inches and feet and  with a negative value such as _-8' 8" _  the value increases by 1 foot each time I click into or out of the value field.  I'm on 0.20.2 but this was happening in previous 0.20 releases as well.  If the value starts at -8' 8" clicking once on the value changes it to -7' 8" then clicking away changes it again to -6' 8" .  The value will continue to change only by clicking in and out of the value field until it reaches -8"

This obviously causes very high risk for adding errors to drawings as a simple accidental click in a field moves objects and changes dimensions of all types.

I did search and wasn't able to find an existing report but maybe I missed it as this issue is 100% repeatable for me.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-01:
Please test it with the current 0.21, there were many improvements it that area.

smrruby, 2023-07-02:
This appears fixed in in the 2023-06-30 build of 0.21

closed 2023-07-02 by smrruby

grubertechnic, 2023-07-02:
Thanks for testing.