FreeCAD/FreeCAD #9862: [Problem] Some error messages don't point to the problem source (e.g. Unit mismatch)

YosypC, 2023-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
```

### Subproject(s) affected?

None

### Problem description

[image]

I've got like 10 to 15 function constraints in a single sketch, and this message doesn't point to any of them. Fortunately, after a bit of thinking, I ended up discovering where the problem was, but doesn't the program already now?

### Anything else?

[MotherboardPlate.zip](https://github.com/FreeCAD/FreeCAD/files/11930304/MotherboardPlate.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-02:
There are files missing while loading.

Components.FCStd
Master.FCStd

YosypC, 2023-07-02:
Updated with every file of the project

grubertechnic, 2023-07-02:
The message in your Screenshot points to a Sketch (not shown on witch file, i think it is not `MotherboardPlate.FCStd`) with name (not Label) `Sketch` and a unit mismatch. If fails not load complete it is not a DAG, but all files with Sketch are complete loaded and recomputed. When i inspect the file Master and its tables, is see, that one length have no unit. When you add an unit, does it work? Which unit system do you use? I use mm/kg/s, since FreeCAD internal work with mm, it can be that it interprets unitless lengths as mm.
[image]

YosypC, 2023-07-02:
I've already fixed the issue, however, while the error points to .Sketch.Constraints, it fails to tell me **which** constraint the error was in. Apologies for not making it clear from the beginning, thanks for the explanation anyway.

grubertechnic, 2023-07-02:
A sorry, i have overread that. Yes, that is true.

closed 2025-01-25 by github-actions[bot]