FreeCAD/FreeCAD #9864: [Problem] Lags in TaskView, some small like Pad, much bigger for primitives (Part and PartDesign)

grubertechnic, 2023-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32531 (Git)
Build type: Release
Branch: (HEAD detached at 5aeba7a46f)
Hash: 5aeba7a46f1deab2cce855d653cb6a550df4faa3 (commit before the issue)
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

FreeCAD lags in TaskView, some are smaller like Pad, some are very big, like prmitives.

The bisect results, that it was intreduced 98b1cbe3e5414f27f7f22a9f0f7172fef09aeae2, probatly due many regex units strings.

Here a video compare

[2023-07-02 22-54.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/52bb3310-c620-4401-badf-14ac869267b5)

It seems, that not every OS havve the same lag impact.

https://forum.freecad.org/viewtopic.php?t=79154

### Anything else?

Maybe a candidate to fix before 0.21 release?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-02:
Forgot a file:

[test.zip](https://github.com/FreeCAD/FreeCAD/files/11931492/test.zip)

adrianinsaval, 2023-07-03:
confirmed, the lag is associated with the configured numbers of decimals displayed. With a normal amount like 2 or 3 the lag is basically imperceptible but when bumping all the way up to 16 the lag is very noticeable.
@ConnorWorrell got any idea how we can improve performance here?

adrianinsaval, 2023-07-03:
curiously up to around 10 decimals the lag is still not too much but then quickly becomes very noticeable

adrianinsaval, 2023-07-03:
maybe we should use boost regex instead of qregularexpression? @wwmayer I remember we kept boost regex around because std regex had very poor performance, do we know if the same happens with qregularexpression?

ConnorWorrell, 2023-07-03:
I'll have to think on this, the algorithm is O(x^n) so its really bad when an exhaustive search is performed. Boost regex won't help.

The issue is it exhaustively searches the 111 in 111mm for a ^ or /.

[image]

adrianinsaval, 2023-07-03:
it seems decimals are supported up to 16 so if boost has good performance up to that point then that may be good enough.

ConnorWorrell, 2023-07-03:
I've got a candidate commit here:
https://github.com/ConnorWorrell/FreeCAD/tree/ExpressionParserRegexOptimization

I measure this branch to be between 15X faster and 10% slower than master, but actual time savings are in the tens of milliseconds range. Its faster, but still noticeable when creating a primitive.

Normally times these short aren't noticeable, but for whatever reason the part design primitive button is dumping hundreds of calls through QuantitySpinbox (attached below, most look like garbage to me), it seems excessive given the dialog only has 9 spinboxes.

| Input                                                                                                     | Previous Time | Updated Time | Speed Percent    | Time Reduction |
|-----------------------------------------------------------------------------------------------------------|---------------|--------------|------------------|----------------|
| 10.x mm                                                                                    | 22190         | 4200         | 528.333333333333 | 0.01799        |
| 10.y mm | 35599         | 4500         | 791.088888888889 | 0.031099       |
| 20.x/(10.x/(10.x/10.xmm))                     | 4478          | 4956         | 90.3551251008878 | -0.000478      |
| acos(Spreadsheet.something*10.xmm)                                                         | 34698         | 2308         | 1503.37954939341 | 0.03239        |

This regex is slightly more aggressive than previously, but outputs are functionally identical.

| Input                           | Output                              |
|---------------------------------|-------------------------------------|
| 1/2                             | 1/2*(1mm)                           |
| 1/2 in                          | 1/2 *(1in)                          |
| 1/2"                            | 1/2*(1\")                           |
| 1 + 1 in                        | 1 *(1mm)+ 1 *(1in)                  |
| 1/(1in+1)                       | 1/(1in+1)                           |
| 3 in + 6 mm                     | 3 *(1in) + 6 *(1mm)                 |
| 1/(2*2mm) + 1/2in + 2^(2+1thou) | 1/(2\*2mm) + 1/2\*(1in) + 2^(2+1thou) |
| 1mm/2in + 1/2mm                 | 1mm/2in + 1/2*(1mm)                 |
| cos(1mm)                        | cos(1mm)                            |
| 1/(2mm)                         | 1/(2mm)                             |
| Parameters.Area/(2*5mm)         | Parameters.Area/(2*5mm)             |
| acos(Spreadsheet.something*1mm) | acos(Spreadsheet.something*1mm)     |
| cos(1) in                       | cos(1) *(1in)                       |
| 1/(1/(1/2mm))                   | 1/(1/(1/2*(1mm)))                   |
| ( 1 + 1 ) kg                    | ( 1 + 1 ) *(1kg)                    |
| 1^2in                           | 1^2*(1in)                           |

<details>
  <summary>Hundreds of calls through spinbox when adding a primitive</summary>

```
23:31:26  Duration 21690
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 21396
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 21771
23:31:26  Copy "10.0000000000000000 mm"
23:31:26  Duration 21767
23:31:26  Copy "10.0000000000000000 mm"
23:31:26  Duration 23097
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 22149
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 21705
23:31:26  Copy "10.0000000000000000 mm"
23:31:26  Duration 21997
23:31:26  Copy "10.0000000000000000 mm"
23:31:26  Duration 21954
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 22094
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 21514
23:31:26  Copy "10.0000000000000000 mm"
23:31:26  Duration 21653
23:31:26  Copy "10.0000000000000000 mm"
23:31:26  Duration 22083
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 21830
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 22157
23:31:26  Copy "10.0000000000000000 mm"
23:31:26  Duration 21910
[... 435 lines cut ...]
23:31:26  Duration 22231
23:31:26  Copy "0.0000000000000000 °"
23:31:26  Duration 22312
23:31:26  Copy "0.0000000000000000 °"
23:31:26  Duration 22388
23:31:26  Copy "0.0000000000000000 °"
23:31:26  Duration 22196
23:31:26  Copy "0.0000000000000000 °"
23:31:26  Duration 22192
23:31:26  Copy "0.0000000000000000 °"
23:31:26  Duration 22331
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 22312
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 22140
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 22593
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 22178
23:31:26  Copy "0.0000000000000000 mm"
23:31:26  Duration 22333
23:31:26  Copy "0.0000000000000000 mm"
```

</details>

ConnorWorrell, 2023-08-04:
The primitive dialog is much slower than the standard dialog because it loads spinboxes for all primitive inputs, even if they are hidden, seems like fixing this would be a lot of effort.

src/Mod/PartDesign/Gui/TaskPrimitiveParameters.ui

closed 2023-08-08 by adrianinsaval