FreeCAD/FreeCAD #9871: [Problem] Fillet Not Created: Recompute Failed!

Electro707, 2023-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git) AppImage
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.56a
  * fasteners 0.3.50
  * Manipulator 1.4.9
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

On `Body`, if a 1mm Fillet is attempted to be created on the highlighted edge, it will only respond with "Recompute Failed". A fillet is not computed until it is 5mm or greater.
[image]

Another part `Body001` is exactly the same as `Body` but the side wall is on the other side. On that part, the maximum fillet size is ~1.3mm before it fails, which is odd that both similar bodies would fail opposite.

### Anything else?

File is attached below:

[Bench_Hub_Mount.tar.gz](https://github.com/FreeCAD/FreeCAD/files/11944134/Bench_Hub_Mount.tar.gz)

Also tested with the latest FreeCAD Stable, both the AppImage and ArchLinux distribution.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

M4xTheMan, 2023-07-04:
I'm on my mobile, therefore I'm not able to check your FreeCAD file but could you create a thread at the forum? Maybe we can sort it out over there: https://forum.freecad.org/viewforum.php?f=3

FlachyJoe, 2023-07-04:
@Electro707 solved by activating the refine property of the operation above fillet.

grubertechnic, 2023-07-04:
I second, that nothing is refined here. Fillets only works, when there are no conflicting geometries stand on its way. Should be closed.

grubertechnic, 2023-07-04:
@FlachyJoe : It can make in very rare cases issues. (in my opinion negligible, i saw then in real life  ~ 1 in a year and to be honest, only in bug reports, never myself, i only remember one or two, but can no find it anymore). I would active auto refine as default, but some do not like it. I have auto refine activated.

Maybe we should rethink that, especially when the TNP is solved. There are already some settings in question (example: single solid in body rule).

At least for this ticket it is meaningless.

M4xTheMan, 2023-07-04:
The common recommendation at the FC forum is to **not** turn it on by default because it can cause unexpected behavior (yes, in very rare cases). But in these cases, the root cause may be easier to recognize / detect.

I'm using it if I have to, therefore autorefine is disabled. In the end, it's up to you.

It may be a good idea to reevaluate after merging the TNP stuff, yes.

Electro707, 2023-07-04:
@FlachyJoe Thanks a ton! Refining it works.

Before I close this ticket, should we add/mention it on the *PartDesign Fillet* wiki page? Sort of how *Part Fillet* mentions to refine the shape if the Part Fillet is having issues.

M4xTheMan, 2023-07-05:
There's already some stuff at the FC Wiki but it could be clearer written for a beginner I think.

https://wiki.freecad.org/PartDesign_Fillet
https://wiki.freecad.org/index.php?title=Part_Fillet/en (Notes an application...)

Is there another place where I should've looked? Thinking about it, this is something which should be discussed at the FC forum.

Electro707, 2023-07-05:
Yeah agreed. Closing this Issue for now and will create a new forum thread (tonight when I got to it).

Thanks all for the help and your time!

closed 2023-07-05 by Electro707