FreeCAD/FreeCAD #9872: Make a cavity in a part made from a sketch "wire closed" using symetry

rakforgeron, 2023-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Version 20.2
rev 29177 +426(Git)
Windows 7 version 6.1 64 bit
"Pocket: Resulting shape is not a solid"
```

### Subproject(s) affected?

Sketcher

### Problem description

I made a sketch from part design, it was half of teh final drawing. I closed the drawing to have a "wire closed" and made a symetry.
Aftter I growed it to make a part from the sketch.
On the part done, I wanted to place two areas to remove some material.
Freecad said "Pocket: Resulting shape is not a solid".
Solution :
I solved the problem by not closing the sketch before to make the material growth.
I was abble to make the part as I wanted.
Regards

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-04:
Please add the file, in screenshots we can only guess. And when i see the screenshots, i would guess, that that is is a user error due invalid constraints.

grubertechnic, 2023-07-08:
@luzpaz : since there is no file and the screenshot intends, that it is a user error (bad Sketch), it should be closed in a certain time.

closed 2023-07-09 by luzpaz

grubertechnic, 2023-07-12:
@luzpaz : thanks