FreeCAD/FreeCAD #9885: Box Selection causes freecad to freeze

Limb, 2023-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

When using FreeCAD on Windows 10, choosing Edit -> Box Selection (or invoking via Shift+B) causes the mouse icon to change to the "box selection" icon, but FreeCAD then freezes and can only be closed via force close. No other actions are responded to by the program. FreeCAD goes unresponsive without clicking the mouse to begin a box selection, as soon as "box selection" is chosen from the edit menu, the program freezes.

I have tried v20.2 on Linux, and the issue is not present. I did try both 0.21 weekly build and compiling from git directly, the issue is present on both.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-12:
Since i have no windows, i can not replicate it.

> I did try both 0.21 weekly build and compiling from git directly, the issue is present on both.

Are you sure that is also on the AppImage? How did you compile the source, did you gave it a `Debug` or `Release` flag `CMAKE_BUILD_TYPE` in cmake? Note, there are some really random freezes in `Debug` build type, which are away in `Release` or `RelWithDebInfo`.

Limb, 2023-07-15:
> Are you sure that is also on the AppImage? How did you compile the source, did you gave it a `Debug` or `Release` flag `CMAKE_BUILD_TYPE` in cmake? Note, there are some really random freezes in `Debug` build type, which are away in `Release` or `RelWithDebInfo`.

To clarify, the issue was not present on the Linux build I tried, the issue is only present on Windows. So I don't believe AppImage is relevant here.

Since I am building for Windows, I don't have a CMAKE_BUILD_TYPE as an option (It was not present on my configuration). I would assume the build type is controlled inside the Visual Studio project. I tried a RELEASE build and the issue was still present.

> May be related to #9386 ? Since you mention this happening exclusively on windows, the first suspect is graphics drivers. Please make sure that your drivers are up to date. If you have both a dedicated and integrated GPU try forcing FreeCAD to run on one of them (test both).

#9386 said the issue was resolved in 0.21, so I was hesitant to reference that issue here. It also happens for both empty and non-empty projects for me, where as they only reported it for empty projects.

Graphics drivers are latest. Issue was present when I tried with iGPU.

> Something else that had caused unexpected freezes in the past are some windows accessibility featuers, can you check your windows setting if you have any accessibility feature enabled?

No accessibility options are enabled.

When I try and step through the code in Visual Studio debug mode, the code goes into the QT libs and before it comes back to traceable code, the issue happens. I am not sure if QT could potentially be the root of the problem, or if the program is just doing a context switch and I miss what is causing the issue during a step over of the code.

adrianinsaval, 2023-07-15:
Qt may be at fault
Can you try enabling software opengl and restart freecad?

If it doesn't work please also try running with a clean config (remove/rename %APPDATA%/FreeCAD)

luzpaz, 2023-07-18:
@Limb :point_up_2:

Limb, 2023-07-18:
> Qt may be at fault Can you try enabling software opengl and restart freecad?

Turning on software OpenGL did resolve the issue. No hangups when using box selection in both 0.20.2 and 0.21 develop branch.

adrianinsaval, 2023-07-19:
Then it is an opengl/graphics driver issue, I suggest you do something like described here: https://forum.freecad.org/viewtopic.php?p=693564#p693564 rather than using software opengl full time, the graphics performance is usually significantly affected by using software opengl.

luzpaz, 2023-09-10:
No response from OP. Closing.

luzpaz, 2023-11-08:
@TimGoll ping

sqwishy, 2023-12-21:
> @TimGoll thanks for reporting, please edit the above post with your full [About](https://wiki.freecad.org/About) info. TIA
> 
> Any way you can run the from the command line and see what error may be generated when it freezes ?
> 
> Would you also check if this can be reproduced on latest windows development build (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) ?

Not OP but, for what it's worth, enabling software OpenGL did not help in 0.21.1, box selection still caused a hang, but software OpenGL did help with "weekly builds 35414-2023-12-18" 759930e. (weekly build still crashed with hardware acceleration iirc.) This was on Windows 11 using an AMD 7000 series GPU.

unixpoint, 2024-03-24:
Not OP either, but wanted to update that the problem also exists on Linux with 0.21.2 on Ubuntu 22.04LTS.

maxwxyz, 2024-04-15:
Is this still present in the current weekly builds?

foresterr, 2024-04-20:
FWIW, I have the same issue in 0.21.2 on Windows, but in my case either turning on software Open GL or using GLonD3D12 make it go away.

maxwxyz, 2024-06-13:
is this still relevant?

foresterr, 2024-06-13:
In my case, box selection does not seem to cause any crashes anymore on the most recent weekly. Software OpenGL is turned off.

closed 2024-06-14 by maxwxyz