FreeCAD/FreeCAD #9887: Get an error on opening file, that has just been created

berebin, 2023-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

Arch

### Problem description

I've done some simple operations in FreeCAD, saved a file and when I try to open it, I get these errors in Report view panel:

`15:37:31  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1600, in attach
    image = Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
<class 'KeyError'>: ('simple',)
15:37:31  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1694, in setDisplayMode
    self.fset.coordIndex.deleteValues(0)
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
15:37:31  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1694, in setDisplayMode
    self.fset.coordIndex.deleteValues(0)
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
15:37:31  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftviewproviders/view_wire.py", line 96, in attach
    self.symbol = gui_utils.dim_symbol()
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftutils/gui_utils.py", line 225, in dim_symbol
    marker = coin.SoSphere()
<class 'NameError'>: name 'coin' is not defined
15:37:31  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftviewproviders/view_wire.py", line 133, in onChanged
    if self.symbol:
<class 'AttributeError'>: 'ViewProviderWire' object has no attribute 'symbol'
`

I use FreeCAD on Mac:
MacBook Air M1, 2020
RAM 16 GB
macOS: Ventura 13.4 (22F66)

What I've done:
1. Open FreeCAD.
2. Create new empty document.
3. Select Arch workbench.
4. Use "Polyline" tool to draw a triangle.
5. Select my triangle in Combo view.
6. Change option "Make face" from TRUE to FALSE.
7. Use "Wall" tool to build a wall.
8. Save the file and exit from FreeCAD.
9. Open the file by double clicking on it.
10. Get errors mentioned above.

If I open the file on WIndows, I get no errors.

I attach FreeCAD file to the post.

Note — when an error appears in the Arch workbench the "Wall" tool disappears.

### Anything else?

1. [Screencast is here, (Google Drive)](https://drive.google.com/file/d/1jbGlr4Iy9qwzdKW1lHmHrRqCHmAgsRiy/view) (1m37s)
2. [Forum thread is here](https://forum.freecad.org/viewtopic.php?t=79399).
3. File is attached, [demo.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11990484/demo.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-07-09:
@berebin can you please test on latest development version https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and see if you can reproduce ?

berebin, 2023-07-09:
Hello, @luzpaz !

I ran some more tests according to your comment and the comment [from forum](https://github.com/FreeCAD/FreeCAD/issues/9887) , here are the results:

### 1. FreeCAD 0.20.2, double click on file.
Got errors as pointed out earlier:
```
20:41:45  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1600, in attach
    image = Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
<class 'KeyError'>: ('simple',)
20:41:45  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1694, in setDisplayMode
    self.fset.coordIndex.deleteValues(0)
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
20:41:45  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1694, in setDisplayMode
    self.fset.coordIndex.deleteValues(0)
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
20:41:45  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftviewproviders/view_wire.py", line 96, in attach
    self.symbol = gui_utils.dim_symbol()
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftutils/gui_utils.py", line 225, in dim_symbol
    marker = coin.SoSphere()
<class 'NameError'>: name 'coin' is not defined
20:41:45  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftviewproviders/view_wire.py", line 133, in onChanged
    if self.symbol:
<class 'AttributeError'>: 'ViewProviderWire' object has no attribute 'symbol'
```
Executed commands in console:
```
>>> print(App.svgpatterns.keys())
dict_keys(['earth', 'general_steel', 'aluminium', 'hbone', 'plastic', 'titanium', 'plus', 'solid', 'steel', 'brick01', 'glass', 'zinc', 'diagonal1', 'diagonal2', 'cuprous'])
>>> print(App.GuiUp)
1
>>> 
```

### 2. FreeCAD 0.20.2, opened file via FreeCAD's "File -> open" menu (FreeCAD was opened before by double clicking on file, see step 1).
Got errors as in the step 1.

### 3. FreeCAD 0.20.2. Started FreeCAD normally and then open file via "File -> open" menu.
Got NO errors, everything was ok.

### 4. FreeCAD 0.20.2. Started FreeCAD normally and then double click on file.
Got NO errors, everything was ok.

### 5. FreeCAD 0.21.0, double click on file.
Got errors as pointed out earlier and some extra errors (about BRep). The object didn’t appear on the screen.

```
20:44:34  Populating font family aliases took 47 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
20:44:35  Failed to load BRep file PartShape.brp
20:44:35  Failed to load BRep file PartShape1.brp
20:44:35  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1620, in attach
    image = Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
<class 'KeyError'>: ('simple',)
20:44:35  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1714, in setDisplayMode
    self.fset.coordIndex.deleteValues(0)
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
20:44:35  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/ArchWall.py", line 1714, in setDisplayMode
    self.fset.coordIndex.deleteValues(0)
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
20:44:35  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftviewproviders/view_wire.py", line 96, in attach
    self.symbol = gui_utils.dim_symbol()
[... 36 lines cut ...]
20:45:14  Unknown command 'Arch_ToggleIfcBrepFlag'
20:45:14  Unknown command 'Arch_IfcSpreadsheet'
20:45:14  Unknown command 'Arch_ToggleSubs'
20:45:14  Unknown command 'Arch_Wall'
20:45:14  Unknown command 'Arch_Add'
20:45:14  Unknown command 'Arch_Remove'
20:45:14  Unknown command 'Arch_Survey'
20:45:14  Unknown command 'Draft_Hatch'
```

### 6. FreeCAD 0.21.0, opened file via FreeCAD's "File -> open" menu (FreeCAD was opened before by double clicking on file, see step 5).
Got errors as in the step 5. The object didn’t appear on the screen.

### 7. FreeCAD 0.21.0, started FreeCAD normally and then open file via "File -> open" menu.
Got errors:

```
20:44:34  Populating font family aliases took 47 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
20:44:35  Failed to load BRep file PartShape.brp
20:44:35  Failed to load BRep file PartShape1.brp
```

The object didn’t appear on the screen.

Full screencast [is here](https://drive.google.com/file/d/13I2JBAqtya9nPzQ9Z1N56V0pd8gKfnTK/view) (4m38s).

luzpaz, 2023-07-09:
Thank you for being thorough! Can you also add the [About](https://wiki.freecad.org/About) info for v0.21.x  
TIA!

berebin, 2023-07-09:
Here it is:
[code]
OS: macOS 13.4
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * BIM 2021.12.0
[/code]

luzpaz, 2023-07-09:
@Roy-043 there are some Draft related warnings/errors in here. Can you take a look ?

Roy-043, 2023-07-10:
The Draft and Arch errors are caused by the fact that those workbenches are not loaded for some reason. So this is primarily a Core and OS issue.

yorikvanhavre, 2023-07-10:
> <class 'NameError'>: name 'coin' is not defined

This probably indicates a coin / pivy / OpenGL problem. It would be interesting to try running another OpenGL application (ex. [Blender](https://blender.org)). If it fails to run too, then we know OpenGL is not properly set on your system. However, since FreeCAD itself is running, it's unlikely. More probably pivy (python implementation of coin)  is not running correctly, and therefore python-based workbenches like Arch or Draft are failing.

It might be  a problem with the pivy package that comes bundled with FreeCAD, but it's strange, we would have heard other Mac users report problems...

berebin, 2023-07-11:
Also I tried to launch Blender — everything is ok. Run Blender, open "Scripting" tab to see the console — there was no any errors in it.
```
PYTHON INTERACTIVE CONSOLE 3.10.12 (main, Jun 20 2023, 20:10:37) [Clang 14.0.3 (clang-1403.0.22.14.1)]

Builtin Modules:       bpy, bpy.data, bpy.ops, bpy.props, bpy.types, bpy.context, bpy.utils, bgl, gpu, blf, mathutils
Convenience Imports:   from mathutils import *; from math import *
Convenience Variables: C = bpy.context, D = bpy.data

>>> print("Hello, world!")
Hello, world!

>>> 
```

yorikvanhavre, 2023-07-12:
ok then your openGL setup is working fine.
Could you try this in the FreeCAD python console?

```python
from pivy import coin
cube = coin.SoCube()
sg = Gui.ActiveDocument.ActiveView.getSceneGraph()
sg.addChild(cube)
```

berebin, 2023-07-12:
Got no error in both versions.

0.20.2:
```
Python 3.10.8 | packaged by conda-forge | (main, Nov 22 2022, 08:44:58) [Clang 14.0.6 ] on darwin
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> FreeCAD.openDocument('/Users/alexey/Desktop/FreeCAD problem/demo.FCStd')
>>> Gui.runCommand('Std_OrthographicCamera',1)
>>> Gui.runCommand('Std_Workbench',19)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> from pivy import coin
>>> cube = coin.SoCube()
>>> sg = Gui.ActiveDocument.ActiveView.getSceneGraph()
>>> sg.addChild(cube)
>>> 
```
0.21.0:
```
Python 3.10.12 | packaged by conda-forge | (main, Jun 23 2023, 22:44:05) [Clang 15.0.7 ] on darwin
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> FreeCAD.openDocument('/Users/alexey/Desktop/FreeCAD problem/demo.FCStd')
>>> # Gui.runCommand('Std_Workbench',15)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> # Gui.runCommand('Std_OrthographicCamera',1)
>>> from pivy import coin
>>> cube = coin.SoCube()
>>> sg = Gui.ActiveDocument.ActiveView.getSceneGraph()
>>> sg.addChild(cube)
>>> # Gui.runCommand('Std_About',0)
>>> 
```

marlanw, 2023-07-13:
Team, I am a VERY junior Freecad user and a MacUser and I have the same issue. I am only using Part Workbench and Part Design Workbench. If I start FCD and open->file (or recent) everything works great. When I click on the .FCSTD file in MacOS file manager, the spashscreen appears and then quits and displays the normal apple dialog box (FreeCAD-0.20.2-29603 quit unexpectedly. )

I'm on MacOS 13.4.1 on an M2 13" Laptop. 

I'm not much of a SW developer, but if there is any data this mere mortal could provide that would help you, happy to do it! 

BTW, I graduated from TinkerCAD and paying kids on Fiverr to do 3D for my Prusa printer. Your SW almost caused me tears when I created my first parameterized design. I am now a FREE man! Thank you (and yes I donated and will probably do more as I create more designs). Love it!

bdieterm, 2023-07-14:
Hi everyone,
just moved over from the forum thread to not split the discussion anymore (user bdm in the forum).
I am only helping to solve the problem/riddle -- I even do not have an Apple computer, so I cannot reproduce the problem.

@yorikvanhavre 
I also took that path with the missing texture resource file.
Your last question is actually already answered.
From the forum thread:
This video at 33 seconds:
https://drive.google.com/file/d/13I2JBAqtya9nPzQ9Z1N56V0pd8gKfnTK/view?usp=sharing

So executing `print(App.svgpatterns.keys())` results in `dict_keys(['earth', 'general_steel', 'aluminium', 'hbone', 'plastic', 'titanium', 'plus', 'solid', 'steel', 'brick01', 'glass', 'zinc', 'diagonal1', 'diagonal2', 'cuprous'])`.

Some entries are missing: {'line', 'simple', 'hatch45R', 'wood', 'cross', 'square', 'woodgrain', 'hatch45L', 'concrete'}
These missing entries are normally all loaded [here](https://github.com/FreeCAD/FreeCAD/blob/10e36fcec6de89491bd36b2c604ed840d1d5a995/src/Mod/Draft/draftutils/utils.py#L739) in function `load_svg_patterns` and assigned [here](https://github.com/FreeCAD/FreeCAD/blob/10e36fcec6de89491bd36b2c604ed840d1d5a995/src/Mod/Draft/draftutils/utils.py#L743). Only the svg files with a "<pattern ...>" tag are loaded in this loop.

`App.svgpatterns` is assigned in python function `load_svg_patterns` (see [here](https://github.com/FreeCAD/FreeCAD/blob/10e36fcec6de89491bd36b2c604ed840d1d5a995/src/Mod/Draft/draftutils/utils.py#L731)). There is no other place where this variable is created. So we know, that the execution of that function was at least started but some svg files from the resources are missing.

The code part `Draft.svgpatterns()` is just returning the dict `App.svgpatterns` (see [here](https://github.com/FreeCAD/FreeCAD/blob/10e36fcec6de89491bd36b2c604ed840d1d5a995/src/Mod/Draft/draftutils/utils.py#L795) and [here](https://github.com/FreeCAD/FreeCAD/blob/10e36fcec6de89491bd36b2c604ed840d1d5a995/src/Mod/Draft/draftutils/utils.py#L776)).
So Yorik, your code will most probably result in a KeyError because of missing entry `simple`.

I was wondering what could be the reason for the missing files.
The missing resource files are theoretically only available after `import Draft_rc` is executed. But this is already done during InitGui (see [here](https://github.com/FreeCAD/FreeCAD/blob/10e36fcec6de89491bd36b2c604ed840d1d5a995/src/Main/FreeCADGuiPy.cpp#L319), [here](https://github.com/FreeCAD/FreeCAD/blob/10e36fcec6de89491bd36b2c604ed840d1d5a995/src/Gui/FreeCADGuiInit.py#L136) and [here](https://github.com/FreeCAD/FreeCAD/blob/10e36fcec6de89491bd36b2c604ed840d1d5a995/src/Mod/Draft/InitGui.py#L179)) and this seems to execute normally without any error.

marlanw, 2023-07-15:
> @marlanw Any error message? Can you try launching FreeCAD from a terminal? Maybe also try the steps above to discard a possible OpenGL problem
> 
> @berebin Thanks for following up. So there is no problem with the OpenGL/coin/pivy setup. Sorry, looking again at your messages that was obvious because you were able to draw polylines, wall, etc. Anyway it's always good to make sure.
> 
> Could you try this in the python console of FreeCAD? This seems to be the culprit, given your error messages...
> 
> ```python
> import Draft
> Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
> ```

Hello Yorik - the terminal "open" command is pretty boring on the Mac so I didn't get much. HOWEVER, I did notice that only my files with "ShapeText" in them caused the error. To be clear, the error is this - when clicking on the FCStd file - FreeCAD appears to start normally, then quits silently - I believe coredumping as the screen MacOS produces looks like the core dump screen.

So I started deleting EVERYTHING from my files until I was left with ONLY a ShapeText and not much else. I'll attach that file. When I click on the the file it coredumps. If I remove the ShapeText the core dump goes away. I'll attach those files. I hope that helps!

[Archive.zip](https://github.com/FreeCAD/FreeCAD/files/12062886/Archive.zip)

berebin, 2023-07-17:
Hi, everyone. Sorry for the long reply, I've been off the net.

I've tried this:
```
import Draft
Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
```
as @yorikvanhavre wrote above. And this:
```
import sys
from PySide2 import QtCore # added during the EDIT
print('Draft_rc before:', sys.modules.get('Draft_rc', '<NONE>'))
print('patterns before:', QtCore.QDir(":/patterns").entryList())
import Draft_rc
print('Draft_rc after:', sys.modules.get('Draft_rc', '<NONE>'))
print('patterns after', QtCore.QDir(":/patterns").entryList())
```
as @bdieterm wrote [on forum](https://forum.freecad.org/viewtopic.php?p=692902#p692902).

Traditionally, [screencast is here](https://drive.google.com/file/d/1kucncKqkbxUHICFO6OUZ50JqcnzHxhez/view?usp=drive_link).

Here are the results:

### 0.20.2, double-click on file
```
>>> import Draft
>>> Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
Traceback (most recent call last):
  File "<input>", line 1, in <module>
KeyError: 'simple'
>>> 
```
and
```
>>> import sys
>>> from PySide2 import QtCore # added during the EDIT
>>> print('Draft_rc before:', sys.modules.get('Draft_rc', '<NONE>'))
Draft_rc before: <module 'Draft_rc' from '/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/Draft_rc.py'>
>>> print('patterns before:', QtCore.QDir(":/patterns").entryList())
patterns before: ['aluminium.svg', 'brick01.svg', 'concrete.svg', 'cross.svg', 'cuprous.svg', 'diagonal1.svg', 'diagonal2.svg', 'earth.svg', 'general_steel.svg', 'glass.svg', 'hatch45L.svg', 'hatch45R.svg', 'hbone.svg', 'line.svg', 'plastic.svg', 'plus.svg', 'simple.svg', 'solid.svg', 'square.svg', 'steel.svg', 'titanium.svg', 'wood.svg', 'woodgrain.svg', 'zinc.svg']
>>> import Draft_rc
>>> print('Draft_rc after:', sys.modules.get('Draft_rc', '<NONE>'))
Draft_rc after: <module 'Draft_rc' from '/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/Draft_rc.py'>
>>> print('patterns after', QtCore.QDir(":/patterns").entryList())
patterns after ['aluminium.svg', 'brick01.svg', 'concrete.svg', 'cross.svg', 'cuprous.svg', 'diagonal1.svg', 'diagonal2.svg', 'earth.svg', 'general_steel.svg', 'glass.svg', 'hatch45L.svg', 'hatch45R.svg', 'hbone.svg', 'line.svg', 'plastic.svg', 'plus.svg', 'simple.svg', 'solid.svg', 'square.svg', 'steel.svg', 'titanium.svg', 'wood.svg', 'woodgrain.svg', 'zinc.svg']
```

### 0.20.2, load FreeCAD first
```
>>> import Draft
>>> Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
<pivy.coin.SoSFImage; proxy of <Swig Object of type 'SoSFImage *' at 0x14d7c4cf0> >
>>>
```
and
```
>>> import sys
>>> from PySide2 import QtCore # added during the EDIT
>>> print('Draft_rc before:', sys.modules.get('Draft_rc', '<NONE>'))
Draft_rc before: <module 'Draft_rc' from '/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/Draft_rc.py'>
>>> print('patterns before:', QtCore.QDir(":/patterns").entryList())
patterns before: ['aluminium.svg', 'brick01.svg', 'concrete.svg', 'cross.svg', 'cuprous.svg', 'diagonal1.svg', 'diagonal2.svg', 'earth.svg', 'general_steel.svg', 'glass.svg', 'hatch45L.svg', 'hatch45R.svg', 'hbone.svg', 'line.svg', 'plastic.svg', 'plus.svg', 'simple.svg', 'solid.svg', 'square.svg', 'steel.svg', 'titanium.svg', 'wood.svg', 'woodgrain.svg', 'zinc.svg']
>>> import Draft_rc
>>> print('Draft_rc after:', sys.modules.get('Draft_rc', '<NONE>'))
Draft_rc after: <module 'Draft_rc' from '/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/Draft_rc.py'>
>>> print('patterns after', QtCore.QDir(":/patterns").entryList())
patterns after ['aluminium.svg', 'brick01.svg', 'concrete.svg', 'cross.svg', 'cuprous.svg', 'diagonal1.svg', 'diagonal2.svg', 'earth.svg', 'general_steel.svg', 'glass.svg', 'hatch45L.svg', 'hatch45R.svg', 'hbone.svg', 'line.svg', 'plastic.svg', 'plus.svg', 'simple.svg', 'solid.svg', 'square.svg', 'steel.svg', 'titanium.svg', 'wood.svg', 'woodgrain.svg', 'zinc.svg']
>>> 
```

[... 22 lines cut ...]
>>> 
```

### 0.21.0, load FreeCAD first
```
>>> import Draft
>>> Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
<pivy.coin.SoSFImage; proxy of <Swig Object of type 'SoSFImage *' at 0x1180aa580> >
>>> 
```
and
```
>>> import sys
>>> from PySide2 import QtCore # added during the EDIT
>>> print('Draft_rc before:', sys.modules.get('Draft_rc', '<NONE>'))
Draft_rc before: <module 'Draft_rc' from '/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/Draft_rc.py'>
>>> print('patterns before:', QtCore.QDir(":/patterns").entryList())
patterns before: ['aluminium.svg', 'brick01.svg', 'concrete.svg', 'cross.svg', 'cuprous.svg', 'diagonal1.svg', 'diagonal2.svg', 'earth.svg', 'general_steel.svg', 'glass.svg', 'hatch45L.svg', 'hatch45R.svg', 'hbone.svg', 'line.svg', 'plastic.svg', 'plus.svg', 'simple.svg', 'solid.svg', 'square.svg', 'steel.svg', 'titanium.svg', 'wood.svg', 'woodgrain.svg', 'zinc.svg']
>>> import Draft_rc
>>> print('Draft_rc after:', sys.modules.get('Draft_rc', '<NONE>'))
Draft_rc after: <module 'Draft_rc' from '/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/Draft_rc.py'>
>>> print('patterns after', QtCore.QDir(":/patterns").entryList())
patterns after ['aluminium.svg', 'brick01.svg', 'concrete.svg', 'cross.svg', 'cuprous.svg', 'diagonal1.svg', 'diagonal2.svg', 'earth.svg', 'general_steel.svg', 'glass.svg', 'hatch45L.svg', 'hatch45R.svg', 'hbone.svg', 'line.svg', 'plastic.svg', 'plus.svg', 'simple.svg', 'solid.svg', 'square.svg', 'steel.svg', 'titanium.svg', 'wood.svg', 'woodgrain.svg', 'zinc.svg']
>>> 
```

berebin, 2023-07-18:
@bdieterm , done in both versions (0.20.2 and 0.21.0) and in both variants (double-click on file and launch FreeCAD first), the result is exactly the same. Since there are more than 100 lines, I am attaching it as a file — [result.txt](https://github.com/FreeCAD/FreeCAD/files/12077273/result.txt)

bdieterm, 2023-07-18:
It seems that `load_svg_patterns` is called when `Draft_rc` was not yet imported.
I tried to replicate the errors on my computer running Linux. The only way to do that without modifying FreeCAD was to create a corrupted single-file workbench in my home directory:

contents of file `/home/bdieterm/.local/share/FreeCAD/Mod/asdf/Init.py`:
```
import sys
if len(sys.argv) > 1:
    App.GuiUp = 0
    import Draft
    Draft.svgpatterns()
```
This resulted in the same behaviour as shown in berebin's videos, but this is surely not the only way that leads to these errors.

@berebin:
Can you please run the following code and report back -- text output, as you did with result.txt, is sufficient.
```
import sys
print('argv:', sys.argv)
import Draft
print('simple found 1:', 'simple' in Draft.svg_patterns())
del App.svgpatterns  # force a reload of the patterns
print('simple found 2:', 'simple' in Draft.svg_patterns())
print('ModDirs:', App.__ModDirs__)
Draft.loadTexture(Draft.svg_patterns()['simple'][1], 128)
```

@berebin: As far as I have seen in the video, you have the BIM workbench installed that is not included in the default installation.
If you feel comfortable with renaming the mod directory for a short test (and then renaming it back), then you could also try this:
rename folder `/Users/alexey/Library/Application Support/FreeCAD/Mod` to `/Users/alexey/Library/Application Support/FreeCAD/Mod_disabled`
then check if you can open the files both ways again (double clicking before FreeCAD was started, and opening after FreeCAD was started) and if there is a different behaviour
finally rename the folder to the original name

bdieterm, 2023-07-22:
The following text from berebin's result-2023-07-21.txt
```
=== FreeCAD 0.20.2, double click on file ===

>>> import sys
>>> print('argv:', sys.argv)
argv: ['/Applications/FreeCAD.app/Contents/Resources/bin/freecad']
```
made me realize, that macOS does not always use command line arguments to pass the filepath.
For `.app` bundles with an "Info.plist" file, the app will receive a special QFileOpenEvent event, see [here](https://stackoverflow.com/a/73577429) and [here](https://doc.qt.io/qt-5/qfileopenevent.html).
FreeCAD receives and processes these events [here](https://github.com/FreeCAD/FreeCAD/blob/b1e521743b359ce02ca8526efbed2feb6331dd4f/src/Gui/GuiApplication.cpp#L152).

How could this cause the errors?

First I show some notes of how files are opened when passed as command line arguments:
```
>> src/Main/MainGui.cpp, int main( int argc, char ** argv )
    ...
    App::Application::Config()["RunMode"] = "Gui";
    ...
    App::Application::init(argc, argv);
    >> src/App/Application.cpp, void Application::init(int argc, char ** argv)
        ...
        initConfig(argc,argv);  --> intializes Python
        initApplication();
        >> src/App/Application.cpp, void Application::initApplication()
        Base::Interpreter().runString(Base::ScriptFactory().ProduceScript("FreeCADInit")); --> runs FreeCADInit.py
        ...
    ...
    Gui::Application::runApplication();
    >> src/Gui/Application.cpp, void Application::runApplication()
        ...
        GUISingleApplication mainApp(argc, App::Application::GetARGV());
        ...
        initOpenInventor();
        ...
        runInitGuiScript();
        >> src/Gui/Application.cpp, void Application::runInitGuiScript()
            Base::Interpreter().runString(Base::ScriptFactory().ProduceScript("FreeCADGuiInit")); --> runs FreeCADGuiInit.py
        ...
        // gets called once we start the event loop
        QTimer::singleShot(0, &mw, SLOT(delayedStartup()));
        ...
        Base::Console().Log("Init: Executing event loop...\n");
        mainApp.exec();
        ...

src/Gui/MainWindow.cpp, void MainWindow::delayedStartup()
    ...
    std::list<std::string> files = App::Application::getCmdLineFiles();
    files = App::Application::processFiles(files);
    >> src/App/Application.cpp, std::list<std::string> Application::processFiles(...)
        ...
        Application::_pcSingleton->openDocument(file.filePath().c_str());
        >> src/App/Application.cpp, Document* Application::openDocument(...)
            ...
            auto docs = openDocuments(filenames, nullptr, nullptr, nullptr, createView);
            >> src/App/Application.cpp, std::vector<Document*> Application::openDocuments(...)
                ...
                auto doc = openDocumentPrivate(path, name.c_str(), label, isMainDoc, createView, std::move(objNames));
                >> src/App/Application.cpp, Document* Application::openDocumentPrivate(const char * FileName, ...)
                    ... reads the file into a new document
    ...
```
That means, files are opened in `delayedStartup()` after `FreeCADInit.py` and `FreeCADGuiInit.py` are executed.

For the .app in macOS, the files are opened in the event handler:
```
>> src/Gui/GuiApplication.cpp, bool GUIApplication::event(QEvent * ev)
    ...
    Application::Instance->open(fn, "FreeCAD");
    >> src/Gui/Application.cpp, void Application::open(const char* FileName, const char* Module)
        ...
        Command::doCommand(Command::App, "FreeCAD.openDocument('%s')", unicodepath.c_str());
        >> [via macro] src/Gui/Command.cpp, void Command::_doCommand(const char *file, int line, DoCmd_Type eType, const char* sCmd, ...)
            ...
            _runCommand(file,line,eType,format.constData());
            >> src/Gui/Command.cpp, void Command::_runCommand(const char *file, int line, DoCmd_Type eType, const char* sCmd)
                ...
                Base::Interpreter().runString(sCmd);
```
There is no instance of the `GUIApplication`. But there is an instance of `GUISingleApplication mainApp` that can be seen in the very first code block. (The heritage order is `QApplication > GUIApplicationNativeEventAware > GUIApplication > GUISingleApplication`.)
So `mainApp` is created between the execution of `FreeCADInit.py` and `FreeGuiCADInit.py`. This should not be a problem normally, because the event loop (`mainApp.exec();`) is started after everything else is finished.
As far as I have seen by using breakpoints, `bool GUIApplication::event(QEvent * ev)` is called several times before the event loop is started with specific events.

This makes me wonder, if this function is maybe also called with a `QFileOpenEvent`.
So I just tried to run the following code block in function `Application::runApplication()` (file src/Gui/Application.cpp).

```
{
    QString file = QString::fromStdString("/filepath/of/demo.FCStd");
    QByteArray fn = file.toUtf8();
    Application::Instance->open(fn, "FreeCAD");
}
```

When I inserted the block before `initOpenInventor();`, then FreeCAD crashed during start-up with a segmentation fault in libCoin.
When I inserted the block after that line and before `runInitGuiScript();`, I had the same behaviour as in berebin's videos.
And when I inserted the block after `runInitGuiScript();`, everything worked normally without any errors.

On my Linux system I cannot dig deeper into this issue -- this will need someone with a FreeCAD build environment on a macOS computer.
I recommend to first check if and why the QFileOpenEvent is handled before `runInitGuiScript();` is executed.
Remember that `QFileOpenEvent` is only issued for `.app` bundles.

I think, the problem could be fixed by adding a single-shot QTimer in the handler for the QFileOpenEvent, similar to `MainWindow::delayedStartup()` so that the file will be opened after the event loop is running (and not before).

luzpaz, 2023-08-23:
bumping for visibility

luzpaz, 2025-01-14:
Is this still relevant ?

marlanw, 2025-01-14:
For me, the error has disappeared with the incredible version 1.

M

berebin, 2025-01-14:
> Is this still relevant ?

In version 1.0.0 the error is gone, everything is fine.

luzpaz, 2025-01-15:
Thanks for reporting back!

closed 2025-01-15 by luzpaz