FreeCAD/FreeCAD #9888: [Problem] SVG rendering of Draft annotations in DraftViews and ArchViews not accurate

Roy-043, 2023-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```
```

### Subproject(s) affected?

Draft

### Problem description

The SVG rendering of Draft annotations in TechDraw DraftViews and TechDraw  ArchViews is not accurate. Note that this SVG rendering is created by Draft code and this is therefore primarily a Draft issue and not a TechDraw issue.

1. Text height is ignored in SVG. See Draft_AngularDims_DraftView.FCStd.
2. Dim arrows are bigger in SVG. See Draft_AngularDims_DraftView.FCStd.
3. Text alignment and position of angular Draft Dimensions is different in SVG. See Draft_AngularDims_DraftView.FCStd.
4. SVG of Draft Labels and Draft Texts that are on the XZ and the YZ plane is faulty. See Draft_Label_ArchView.FCStd.
5. Text alignment and position of linear Draft Dimensions is different in SVG if they are on the XZ and the YZ plane. See Draft_Label_ArchView.FCStd.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=79481

[Draft_DraftView_ArchView.zip](https://github.com/FreeCAD/FreeCAD/files/11990499/Draft_DraftView_ArchView.zip)

Files in ZIP:
* Draft_AngularDims_DraftView.FCStd
* Draft_Label_ArchView.FCStd

![Draft_AngularDims_DraftView](https://github.com/FreeCAD/FreeCAD/assets/70520633/bd25f69c-b969-4aeb-9ed3-7bab2dbbf2f7)
![Draft_Label_3DView](https://github.com/FreeCAD/FreeCAD/assets/70520633/926678ff-5e61-4213-9664-9efa6141bdf1)
![Draft_Label_ArchView](https://github.com/FreeCAD/FreeCAD/assets/70520633/7f6ef3a3-ccac-4435-bc87-64e7f96c46f2)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-07-10:
* About the rotated 90° annotations: There is IIRC in Draft SVG module some code to "straighten up" (by rotating 180°) texts that are detected to be upside down. I believe that's what's happening here. Maybe there should be an option for the user to disable this mechanism...

* The non-XY texts problem is not new either... I think the whole code that projects texts needs to be redone...

Roy-043, 2023-11-08:
@luzpaz You are adding the "Help wanted" label to many issues ATM. What is you intention? In a way all issues are candidates for this label, but it would become meaningless if applied to too many of them.

luzpaz, 2023-11-08:
@Roy-043 Thanks, I'll be more discrete. But I don't think it's a given that any open issue is fair game. There has been more interaction with devs when that label is attached.

Roy-043, 2025-01-07:
This is still an issue.

ErtugrulEge, 2025-09-20:
Just for a note. Preferences->Draft->Texts and Dimensions->Scale multiplier doesn't work in Techdraw page