FreeCAD/FreeCAD #9890: Freecad recompute fails when adding two rectangles to initial pad sketch

waroberts, 2023-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * fasteners 0.4.6
  * Assembly4 0.12.4
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

I have a sketch of an instrument panel layout with a series of 5 round holes and two rectangles with screw holes above and below.  I pad this out to 0.090" and works find.  Then I map to the rear face of the Pad a rectangle slightly smaller and pad 5 mm.  This works.  The next steps are to cut a central pocket in the second pad to the back face of the panel to expose the holes.  This works.  Next, mapping respectively to the outer surfaces of each of the secondary pad a small pocket to allow a tab to insert and hold this panel face in place.  This was done on apple silicon M2/Ventura and set up fine.  When I added two new rectangular holes, Freecad gets lost and cannot recompute anything other than the original pad on Apple/Intel chips.  It refuses to face map and will not recompute with a failure message saying recompute failed and nothing else.
[screenshot-orig.pdf](https://github.com/FreeCAD/FreeCAD/files/11990687/screenshot-orig.pdf)

Adding a rectangular cut out to the original sketch demonstrates pad working properly here.
[added-rectangle.pdf](https://github.com/FreeCAD/FreeCAD/files/11990694/added-rectangle.pdf)

At this point, the remainder of the pocket series and the secondary pad fails with a complaint of 

_07:45:37  Traceback (most recent call last):
  File "<string>", line 2, in <module>
<class 'Part.OCCError'>: Both points are equal
07:45:37  Failed to add box: Both points are equal_ 

[reportview.pdf](https://github.com/FreeCAD/FreeCAD/files/11990699/reportview.pdf)

I'll try to make the mod on the Apple silicon later today, if I have time and see what happens.  This operating system/hardware info is
macOSX Big Sur v 11.7.3
MacBook Pro (Retina 15-inch Late 2013)
Processor 2.6 GHz 1600 MHz DDR3
Graphics: NVIDIA GeForce GT 750M 2 Gb

I will make the .FCStd file available if needed.  Thanks.

### Anything else?

[EABPanel-copy.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11990708/EABPanel-copy.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-08:
Duplicate of #8432

@luzpaz : can be closed

adrianinsaval, 2023-07-13:
see https://wiki.freecad.org/Topological_naming_problem

this is a known problem tracked at the issue grubertechnic linked, a solution is underway and expected to be included in the next version after 0.21