FreeCAD/FreeCAD #9893: [Problem] Regression: wrong object placement

PLChris, 2023-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.54
  * sheetmetal 0.2.62
  * ExplodedAssembly
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

None

### Problem description

The Placement of objects can change when loading a file created with a previous version. A description is given in [this forum post](https://forum.freecad.org/viewtopic.php?t=79429). While the Loft in the original file was placed at (0,0,0), no rotation, it changed to z=-27 and a rotation.

[Link to the file](https://forum.freecad.org/download/file.php?id=229384).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lucapalazzo, 2023-07-11:
I can confirm the wrong behaviour with other models.

adrianinsaval, 2023-07-13:
Is this still a problem with the latest build?

adrianinsaval, 2023-07-13:
> I can confirm the wrong behaviour with other models.

please state your full version info as this problem appear to be macOS specific

lucapalazzo, 2023-07-13:
HI @adrianinsaval, last weekly solved the problem.
Thanks
[code]
OS: macOS 13.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33576 (Git)
Build type: Release
Branch: master
Hash: 8e6fbd2886d3ad51b75232bd0019d92d496e80cb
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * Woodworking 0.20.29177
  * Render 2022.4.0
  * WebTools
  * Manipulator 1.5.0
  * kicadStepUpMod 10.19.4
  * fasteners 0.4.52
  * sheetmetal 0.2.60
  * Assembly4 0.50.2
[/code]

chennes, 2023-07-13:
@PLChris can you confirm that this is now fixed for you as well? Thanks!

adrianinsaval, 2023-07-13:
It was most likely just a packaging problem then

adrianinsaval, 2023-07-25:
@PLChris could you confirm this is not a problem anymore?

adrianinsaval, 2023-08-31:
there was a report of this being fixed and no further reports of the problem so I'm closing the issue, feel free to comment or reopen if you experience this again