FreeCAD/FreeCAD #9897: [Problem] edit->Preferences->general deselect the Enable splash screen at start up radio box, FreeCAD cannot be opened

Lian-Zekun, 2023-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
```

### Subproject(s) affected?

None

### Problem description

[image]
Deselecting causes FreeCAD to fail to open again

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-07-09:
Please confirm on development version (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Lian-Zekun, 2023-07-10:
The FreeCAD-0.20.2-WIN-x64-installer-3.exe file I downloaded from this website https://www.freecad.org/downloads.php Can reproduce the problem I described.
I downloaded the installation package for this website and found that the issue has been resolved https://github.com/FreeCAD/FreeCAD/releases/tag/0.20.2.

Lian-Zekun, 2023-07-10:
> Please confirm on development version (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

The FreeCAD-0.20.2-WIN-x64-installer-3.exe file I downloaded from this website https://www.freecad.org/downloads.php Can reproduce the problem I described.
I downloaded the installation package for this website and found that the issue has been resolved https://github.com/FreeCAD/FreeCAD/releases/tag/0.20.2.

yorikvanhavre, 2023-07-10:
Cannot reproduce on Linux. Can another Windows user confirm?

Lian-Zekun, 2023-07-10:
> Cannot reproduce on Linux. Can another Windows user confirm?

I used a Windows computer that had never installed FreeCAD before and could reproduce it.
But since the latest code of 0.20.2 cannot be replicated, there should be no problem.

yorikvanhavre, 2023-07-10:
Ok I'll close the report then!

closed 2023-07-10 by yorikvanhavre