FreeCAD/FreeCAD #9898: [Problem] Using ImportGUI.insert to import STEP returns 'None' instead of the inserted object

innovatingdev, 2023-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Fedora Linux 38 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

When importing a step file within a macro, the method `ImportGui.insert` used to return the imported object in FreeCAD 0.19.3. In FreeCAD 0.20.2  (0.20.1, 0.20) it returns `None`

To test, just create a new document, place any working STEP file in /tmp/test.step, run the following macro:
```
import FreeCAD
import ImportGui

imported = ImportGui.insert(u"/tmp/test.step", "Unnamed")
print(imported)
```
The file will get imported, but it will print e.g. `<Part::PartFeature>` in FreeCAD 0.19.3 and `None` in FreeCAD 0.20.

I am relying on the return value in automation scripts, when e.g. I need to rename and refer to the imported bodies.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bdieterm, 2023-07-09:
Hi @innovatingdev!
To get the imported object as a return value, you have to enable the LinkGroups, e.g. via the menu in FreeCAD:
File -> Preferences -> Import/Export -> STEP -> Use LinkGroup

[source](https://github.com/FreeCAD/FreeCAD/blob/c09ef943f76a9693af6a045220070f685847e458/src/Mod/Import/App/ImportOCAF2.cpp#L577)

luzpaz, 2023-07-09:
Does the macro need to import a specific library ?

innovatingdev, 2023-07-09:
Hey @bdieterm. Actually this worked. Thank you very much.
Btw.: There was not tab 'STEP' in Import/Export. Had to first import one STEP file and the tab appeared.

@luzpaz No. Was just reducing the code to the snippet above to narrow down the issue and to help reproducing. No special library required.

benj5378, 2023-07-22:
@innovatingdev 

To expand on @bdieterm, `ImportGui.insert(u"/tmp/test.step", "Unnamed", useLinkGroup=True)` will also return the object you are looking for. This is a change of the default behaviour and not a bug. @luzpaz could you please close this?

closed 2023-07-23 by luzpaz