FreeCAD/FreeCAD #9899: [Problem] Links to external objects fail to resolve when document is loaded

dajt1725, 2023-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Rocky Linux 8.8 (Green Obsidian) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git) AppImage
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Attached are three documents: Components, 1D1A_Warehouse, and 3D2A_UnderDrinking1.  Both Warehouse and UnderDrinking1 contain links to objects in Components.  If I start FreeCAD, load Warehouse, then attempt to load UnderDrinking1, the links in UnderDrinking1 fail to resolve:
20:22:31  5.61e-07 <App> Document.cpp(2122): _D1A_Warehouse#Link.LinkedObject: Time stamp changed on link Components#Part004
20:22:31  0.0118593 <App> Document.cpp(2122): _D1A_Warehouse#Link001.LinkedObject: Time stamp changed on link Components#Part004
20:22:31  0.0119944 <App> Document.cpp(2122): _D1A_Warehouse#Link002.LinkedObject: Time stamp changed on link Components#Part004
20:22:31  0.0120582 <App> Document.cpp(2122): _D1A_Warehouse#Link003.LinkedObject: Time stamp changed on link Components#Part004
20:22:31  0.0121174 <App> Document.cpp(2122): _D1A_Warehouse#Link008.LinkedObject: Time stamp changed on link Components#Part005
20:22:31  0.037509 <App> Document.cpp(2122): _D1A_Warehouse#Link009.LinkedObject: Time stamp changed on link Components#Part005
20:22:31  0.0376404 <App> Document.cpp(2122): _D1A_Warehouse#Link010.LinkedObject: Time stamp changed on link Components#Part005
20:22:31  0.0377006 <App> Document.cpp(2122): _D1A_Warehouse#Link011.LinkedObject: Time stamp changed on link Components#Part005
20:22:31  0.0377609 <App> Document.cpp(2122): _D1A_Warehouse#Link004.LinkedObject: Time stamp changed on link Components#Part004
20:22:31  0.0378418 <App> Document.cpp(2122): _D1A_Warehouse#Link005.LinkedObject: Time stamp changed on link Components#Part004
20:22:31  0.0379007 <App> Document.cpp(2122): _D1A_Warehouse#Link006.LinkedObject: Time stamp changed on link Components#Part004
20:22:31  0.0379593 <App> Document.cpp(2122): _D1A_Warehouse#Link007.LinkedObject: Time stamp changed on link Components#Part004
20:22:31  0.0380181 <App> Document.cpp(2122): _D1A_Warehouse#Link012.LinkedObject: Time stamp changed on link Components#Part005
20:22:31  0.0380829 <App> Document.cpp(2122): _D1A_Warehouse#Link013.LinkedObject: Time stamp changed on link Components#Part005
20:22:31  0.0381422 <App> Document.cpp(2122): _D1A_Warehouse#Link014.LinkedObject: Time stamp changed on link Components#Part005
20:22:31  0.0382015 <App> Document.cpp(2122): _D1A_Warehouse#Link015.LinkedObject: Time stamp changed on link Components#Part005
20:22:31  0.0382683 <App> Document.cpp(2122): _D1A_Warehouse#Link017.LinkedObject: Time stamp changed on link Components#Part003
20:23:15  44.0756 <App> Document.cpp(2127): _D2A_UnderDrinking1#Link.LinkedObject: Link not restored
Linked object: Part001
Linked document: Components
20:23:15  44.0757 <App> Document.cpp(2127): _D2A_UnderDrinking1#Link001.LinkedObject: Link not restored
Linked object: Part
Linked document: Components
20:23:15  44.0758 <App> Document.cpp(2127): _D2A_UnderDrinking1#Link002.LinkedObject: Link not restored
Linked object: Part002
Linked document: Components
20:23:15  44.0758 <App> Document.cpp(2127): _D2A_UnderDrinking1#Link003.LinkedObject: Link not restored
Linked object: Part002
Linked document: Components
20:23:15  44.0758 <App> Document.cpp(2127): _D2A_UnderDrinking1#Link004.LinkedObject: Link not restored
Linked object: Part007
Linked document: Components
20:23:15  44.0759 <App> Document.cpp(2127): _D2A_UnderDrinking1#Link005.LinkedObject: Link not restored
Linked object: Part009
Linked document: Components
20:23:15  44.0759 <App> Document.cpp(2127): _D2A_UnderDrinking1#Link006.LinkedObject: Link not restored
Linked object: Part006
Linked document: Components
20:23:15  44.0759 <App> Document.cpp(2127): _D2A_UnderDrinking1#Link007.LinkedObject: Link not restored
Linked object: Part013
Linked document: Components

If I manually load Components before loading the other two files, they load successfully.

### Anything else?

[Components.zip](https://github.com/FreeCAD/FreeCAD/files/11995354/Components.zip)
[UnderDrinking.zip](https://github.com/FreeCAD/FreeCAD/files/11995356/UnderDrinking.zip)
[Warehouse.zip](https://github.com/FreeCAD/FreeCAD/files/11995358/Warehouse.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-29:
Perhaps semi-related to #7596 ?

krushia, 2024-05-07:
#13841 Probably the same issue

Zolko-123, 2024-05-07:
Names cannot begin with numbers

nazar-pc, 2024-07-28:
Just hit this with latest dev build, exactly as described:
```
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * parts_library
  * Curves 0.6.40
```

closed 2025-02-06 by github-actions[bot]