FreeCAD/FreeCAD #9903: [Problem] Draft Downgrade of PD Body deletes Body and leave Pad

macdroid53, 2023-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git) AppImage
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Manipulator 1.5.0
  * Defeaturing 1.2.0
  * Help 1.0.3
  * kicadStepUpMod 10.19.4
  * Assembly3 0.12.0
  * Curves 0.6.12
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * fasteners 0.4.56
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * sheetmetal 0.2.63
  * CurvedShapes 1.0.4
  * lattice2 1.0.0
  * QuickMeasure 2022.10.28
[/code]
```

### Subproject(s) affected?

None

### Problem description

This forum thread describes and discusses. ](https://forum.freecad.org/viewtopic.php?t=79539)

In short, a Draft Downgrade of a PD Body leaves the tip feature in the tree. It is devoid of the Body and Origin and in a undefined state for further use.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-07-10:
Not sure what the best path here, deleting the entire body tree seems brutal to me... Maybe for bodies the body should not be deleted? What do you think @Roy-043 ? OTOH Downgrading IS a brutal operation...

Roy-043, 2023-07-11:
I think it is better to keep the standard behavior and delete the source object. At least for the 0.21 dev cycle.

Since deleting any object can break things, maybe we should aim for a generic solution in the next dev cycle. Perhaps prompt the user before deletion (always, or if InLists and OutLists warrant it)?

There is a side-issue that I tried to raise in the Forum topic: What should happen if a feature or sketch of the body is selected? For now I suggest explode that feature and still delete the body.

PR #9934 (2023-07-18): Draft: Improve downgrading of PD bodies
Fixes #9903. See there.

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

---
closed 2023-07-18 by Roy-043