FreeCAD/FreeCAD #9906: Preference not responding

dch1999, 2023-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

This issue is in both window 10 and 11 running on a Ryzen. 12 core process and Radeon rx6650xt graphic card
I click EDIT/PREFERENCES
Try to make a change in preferences and immediately get the blue waiting wheel and Preferences not responding in the window title bar.
If I wait long enough it does come back but still limited or no changing of parameters.
Can't scroll down the page or use the small arrow upper right to access other tabs
Basically, I can't edit the preferences or with very little success.  It times out, window says it is not responding.
the rest of Freecad seems to be working, though I am new to it and have done very little other than to try to change the preferences.  Newly installed about 7/9/23

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-07-13:
Can you try if this is still a problem in the dev release please? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

dch1999, 2023-07-14:
I am new to this, what do I do with the files, there are two for windows a .7z and a .txt file.   Neither seems like an install file….

Dave

luzpaz, 2023-07-15:
7z is a compressed format. You can use https://www.7-zip.org/download.html to decompress it.

adrianinsaval, 2023-07-15:
The executable will be at `bin\FreeCAD.exe` in the extracted folder

benj5378, 2023-07-22:
@dch1999 Did you also experience the error in the latest dev version? If yes, could we get you to run the FreeCAD in a debugger? (we can give more details on how-to if needed)

dch1999, 2023-07-22:
I had an issue with my pc had to rebuild.   I will do as you suggested as soon as my pc is back up and running..  Thanks for the follow up…

Dave

luzpaz, 2023-08-13:
Haven't heard from OP. Closing. Please test on latest stable and if issue persists, please re-open ticket

closed 2023-08-13 by luzpaz