FreeCAD/FreeCAD #9908: Techdraw Rich Text Annotations lost in Export PDF.

JArmannFannberg, 2023-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 13.4
Word size of FreeCAD: 64-bit
Version: 0.21.0.33516 (Git)
Build type: Release
Branch: master
Hash: c09ef943f76a9693af6a045220070f685847e458
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * parts_library
  * A2plus 0.4.60n
  * BIM 2021.12.0
  * Help 1.0.3
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

Having added Rich Text Annotations to a view and executing "file/export pdf", the RTAs disappear from the page window and are not exported with the drawing itself. The RTA items still exist in the tree-view, and appear in the page window as they are clicked-on in the tree-view. 
Leaders seem to survive the operation, as well as "normal" annotation objects. 
Have earlier used the x86 version of Freecad (on the same M1 Mac) without these problems. 
Print (/preview) and then print to pdf (save pdf) does seem to work "correctly" (but then, what about the scale).
Drawing is a view of objects created in Sketcher and BIM. 
Screenshot before export:
[image]
Screenshot after export:
[image]
Resulting PDF file:
[G2-20x-1.pdf](https://github.com/FreeCAD/FreeCAD/files/12017228/G2-20x-1.pdf)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-07-11:
Are we sure the RTAs are missing?  I seems to me that they are there but really small.  

The font-size adjustment for Svg export is being performed for Pdf export, but it should not be.

WandererFan, 2023-07-11:
Any chance you could share the FCStd file?  It tells me much more than the pdf.

JArmannFannberg, 2023-07-11:
The FCStd file is not stand-alone - this is a reconstruction of a 6-floor building, so that I use Arch-external references extensively. The outer periphery of the building is largely the same, but interior is varying. I have had to solve inserting windows in curved walls with some tricks, thus walls are generally in portions. 

There are layers of inheritance in the model: 
- Sketches for periphery and common structures like stairwell.
- Floor, virtual stairwell floor and outer walls in portions. 
- Each floor inherits from these two files and some others as well.
- Often, sketches (that appear as draft lines when used as external Arch-references) are used as external references when making new sketches, or they are converted to sketches and then constrained and perhaps modified. 
- Because of the Arch-reference absolute path, it becomes tedious to move the files to a new path. If I send you a few files, perhaps you can create the original path - and it might just work.
 
It is true that I can export to SVG with the RTAs, but with very small fonts and unproportianate line spacing. In that case, the RTAs do not disappear from the page window, in contrast to the PDF-export.  

Newest observation:
After restart of Freecad (or did it crash while I was out for a refreshing walk?) the behaviour seems to have improved. The RTAs are exported along with other drawing details, and the RTAs are preserved in the page window. But - this is actually the previous version I have been using till now, namely: 

[code]
OS: macOS 13.4
Word size of FreeCAD: 64-bit
Version: 0.21.0.33411 (Git)
Build type: Release
Branch: master
Hash: 127404d793d5c2081383b46a7b677426d9a0c5aa
Python 3.11.4, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * parts_library
  * A2plus 0.4.60n
  * BIM 2021.12.0
  * Help 1.0.3
  * Assembly4 0.50.2
[/code]

So, fortunately I still have this version installed - the new version did not install over it. Suppose the install was incomplete in that respect. I will try to run the newer version again, and check that it is still behaving erraneously.

Johann A. Fannberg.

JArmannFannberg, 2023-07-11:
WandererFan, you are right: 
The RTAs are shown in the exported document. With a font size of 12, the font height is just about the width of a line in the drawing, and the texts are not visible without extensive zoom-in. The same is true for the texts in the page window - they are just as almost non-visible clutter on the screen, but can be seen after repeated zoom-in. 
Then, when clicked-on in the tree, they appear in the correct size. Are they scaled down with the page scale (1/50)?

Johann.

WandererFan, 2023-07-13:
Should be fixed by PR #9917 / commit f83a3fbb6a

JArmannFannberg, 2023-07-13:
Thanks WandererFan, superb response! Feel free to close issue.

closed 2023-07-14 by luzpaz

benj5378, 2023-08-06:
https://bugreports.qt.io/browse/QTBUG-65562?jql=text%20~%20"QGraphicsSvgItem"