FreeCAD/FreeCAD #9911: [Problem] Axis cross affects the object center rotation mode

Rexbas, 2023-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: KDE neon 5.27 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33442 +3 (Git)
Build type: Release
Branch: rotationcenter
Hash: 1ae6a68ee0842d4e35b609556d5105bed3a6125f
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
[/code]

The about page did not give me the right branch name and +n in version
```

### Subproject(s) affected?

None

### Problem description

Axis cross affects the "object center" rotation mode. Steps to reproduce (needs #9909 for visualization):
1. Ensure the axis cross is disabled.
2. Make a simple cube.
3. Set rotation mode to "Object center".
4. Start dragging. With the axis cross disabled, the center should be at the center of the object. This is expected.
5. Enable the axis cross.
6. Start dragging again. Now the rotation center is not shown at the center of the object but depending on the zoom level more towards the center of the axis cross. I would expect that the axis cross does not have an effect on the object center but apparently it does.

https://github.com/FreeCAD/FreeCAD/assets/14298143/8bad2e17-f622-487f-9f72-954dded24b01

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-07-13:
I believe (and I'm sure someone will correct me) that the Axis Cross is different to the _origin coordinates_ and the _nav cube_ in that it's actually a mesh/shape rather than just a drawing/texture hence it affects the 'mass' and centre of rotation.

Now whether the code can be made to exclude the Axis Cross mesh from the total objects in the scene, that's the question to the devs or does that whole way the Axis Cross is created need changing??

Rexbas, 2023-07-16:
The navicube and the axis cross at the bottom right of the screen are drawn using OpenGL calls. The axis cross at the origin is added to the scene graph. The origin is also added to the scene graph. The object center is determined by finding the bounding box center of the scene graph so both the axis cross and origin affect the rotation center.

Maybe we should split the actual object and the 3D UI that is not part of the object into different groups and only take the bounding box center of the object group. I will see if I can make something work.

PR #9969 (2023-08-16): Gui: Improve object center rotation mode
- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
---
This is my attempt to fix #9911. I created a group in `pcViewProviderRoot` called `objectGroup` which contains basically everything that is allowed to affect the rotation center. Things that should not affect the rotation center are directly added as a child of `pcViewProviderRoot`. The axis cross is directly added to `pcViewProviderRoot` and should no longer affect the rotation cen [... cut]
closed 2023-08-16 by chennes