FreeCAD/FreeCAD #9914: [Problem] When I resize the Combo View, it leaves behind a vertical line

K4LCIFER, 2023-07-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.9, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
[/code]
```

### Subproject(s) affected?

None

### Problem description

When I drag the combo view to resize it, it leaves behind a vertical line as shown below:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

K4LCIFER, 2023-07-13:
This appears to be an issue with the "ProDark" theme. When I switch away from this theme to, for example, "Darker-Orange", the problem goes away.

Jamtek, 2023-07-24:
I am seeing this on recent conda build in 0.21 on my M2 MacBook. I also see a horizontal line after resizing report view and or python console.

https://forum.freecad.org/viewtopic.php?t=79877

```
[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33631 (Git)
Build type: Release
Branch: master
Hash: 2167fb6f7ec6ac5c2f6365d4c3035d2db6879b1e
Python 3.11.0, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.58
  * Curves 0.6.12
[/code]
```

luzpaz, 2023-07-25:
@MisterMakerNL is this a phenomena you've encountered too ?

MisterMakerNL, 2023-07-25:
@K4LCIFER Could you test it with this stylesheet:
https://github.com/MisterMakerNL/Freecad-Built-in-themes-beta/blob/main/Dark%20theme%20beta/Test_stylesheet.qss
You can also install beta stylesheets true the addon explorer.

mfro0, 2023-07-26:
Tested with the provided stylesheet - problem still exists (contrary to @K4LCIFER 's statement above, for me the graphics glitch also appears for me with other stylesheets as well),

As soon as you move the vertical divider between the combo view and the view right to it (it doesnt't seem to matter if that is a graphics view or anything else), a line appears that seems to be a leftover of the divider.

Note the vertical line in attached screenshot. It stays at the exact place it appeared at the first combo view resize and doesn't go away even if you move the window border later.

[image]

[edit: I just realised this also happens when resizing any other (also horizontal) divider.

[image]

Might be pure coincidence, but note that I recently ditched Gnome in favour of KDE and I'm unable to tell if I have seen this glitches with the former (might just as well be related to a recent FreeCAD change)

```
OS: Ubuntu 22.04.2 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33639 (Git)
Build type: Release
Branch: master
Hash: 1580fe7c9696654803a93da8cf372f75e75149fa
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.0.dev
Locale: C/Default (C)
Installed mods: 
  * lattice2 1.0.0
  * Assembly4 0.50.2
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * Silk 0.1.3
  * Behave-Dark-Colors 0.1.1
```

yorikvanhavre, 2023-07-26:
I had this problem before too, but with @MisterMakerNL's new versions the problem has disappeared (Debian+Gnome)

mfro0, 2023-07-26:
For me, the problem goes away with "no style sheet" selected from the preferences. So it's probably not related to code.

adrianinsaval, 2023-07-26:
maybe you have leftover settings from gnome that are messing up your system, what qt theme do you have configured in KDE? what is the result of running `echo $QT_QPA_PLATFORMTHEME` in your terminal? Try setting it to "kde" if it isn't already

MisterMakerNL, 2023-07-26:
If you zoom in on the line it shows as a gradient, so I added a color to all gradients to check what gradient is causing it.

There are no borders in the theme with that color so it's a background. But I have some borders thay are set to 0 or transparent I think that maybe is the issue. I'll clean up the test stylesheet from any borders that are not vissible.

MisterMakerNL, 2023-07-26:
I think I found the line in my view, it is replaced in your screenshot by a colored line.
Which is weird.
But the line I see looks to be the line you see floating.
[image]

MisterMakerNL, 2023-07-26:
Could you test the addon, even if it didn't fixed it, it looks a lot cleaner now. I updated all stylesheets.
https://github.com/MisterMakerNL/Freecad-Built-in-themes-beta/commit/ff84c20f3174829b760697717f49136ca762b8a0

mfro0, 2023-07-26:
did you change Test_stylesheet.qss as well? That doesn't seem to change anything. Artifact is still there.

MisterMakerNL, 2023-07-30:
Could you test it again with the addon, and delete the addon or uninstall it instead of updating it. 
Mine also didn't update although it said it did.
It should look like this:
[image]

Jamtek, 2023-08-01:
On my Macbook ARM, I noticed that the lines are not visable if I select 'No Display sheet' or 'Behave-dark' with Classic theme or Light Theme.

I would also note that the Combo view area with no sytlesheet still appears as a dark theme.
[image]

The Task view under the Combo view has a white background and a mixture of white and gray text.
[image]

MisterMakerNL, 2023-08-01:
@Jamtek I don't think those issues are related to this issue. I also have no idea what is going on, so far as I understand Freecad on Linux and Mac follows the system style. That is probably why it is dark.
I think you might need too restart after changing to no-stylesheet.

MisterMakerNL, 2023-08-03:
Bumb can someone test if this issue is still present in the addon version:
https://github.com/MisterMakerNL/Freecad-Built-in-themes-beta

Jamtek, 2023-08-04:
Thanks @MisterMakerNL - Still seeing lines [Freecad-Built-in-themes-beta] M2 Mac, when using 'Dark theme beta, Light theme beta with the assigned stylesheets. If I change to No stylesheet the lines no longer appear.

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Freecad-Built-in-themes-beta 1.2.0
```
[image]

[image]

luzpaz, 2023-10-11:
@Jamtek can you test in dev version to see if issue is still reproducible (but using MisterMakerNL's workabour stylesheets ?

MisterMakerNL, 2023-10-11:
I have not been able too reproduce thi, which makes it very annoying and hard too fix.
But yesterday a new overlay system was pushed, hopefully tackeling this bug in the process.

Jamtek, 2023-10-13:
@luzpaz ,
I am happy to report I am no longer seeing the vertical or horizontal lines when resizing the frames on my M2 Macbook air. Tested the ExtremeProDark, ExtremeProLight and the new OpenDark preference packs. 

```
OS: macOS 14.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34651 (Git)
Build type: Release
Branch: main
Hash: aa5bea4e2e8089bf00f61075a0860dacda42dc8b
Python 3.11.0, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.10.2
  * A2plus 0.4.60n
  * BIM 2021.12.0
  * fasteners 0.4.66
  * ExtremeProDark 2.9.0
  * ExtremeProLight 1.6.0
  * Curves 0.6.13
```
Thanks @MisterMakerNL!

luzpaz, 2023-10-13:
@K4LCIFER @mfro0 can y'all reproduce on latest dev version ?

luzpaz, 2023-10-13:
Closed #8197 as a dupe of this more active ticket.

luzpaz, 2023-10-13:
Closing this ticket as well unless any one else can reproduce. If so, please re-open ticket.

closed 2023-10-13 by luzpaz

X-Ryl669, 2023-10-19:
Probably also linked, right button menu background is transparent and this is really disturbing.
[image]

MisterMakerNL, 2023-10-19:
Extreme pro is outdated and I am going to remove it. It's replaced by modern style in freecad.
Latetst version has a complete new interface with the ovetlay, changes this still in are very small. Although if someone her is able to test the latest dev version to test this bug that would be nice!

X-Ryl669, 2023-10-19:
I am able to test it, if only I knew what and how to test it. 

I've used the addon manager in Freecad to update the theme then cycled through all dark version of them and none worked (see above for the list of themes I've tried). What should I change to test the last version so it's useful for you to debug?

MisterMakerNL, 2023-10-19:
@X-Ryl669 I think the weekly build has the overlay:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Not sure if it got my latest PR for the theme's.

obelisk79, 2023-10-20:
@X-Ryl669 is the behavior also present with OpenDark (available in addon manager)?

X-Ryl669, 2023-10-20:
The weekly build doesn't have any issue (0.22dev), so I guess it's not worth fixing it for 0.21 if it's already going away in next release.

I haven't tried OpenDark.

morphlaugh, 2024-02-02:
@obelisk79 I am seeing the same issues he reported on 0.21.2 for Mac Intel.  The problem goes away with OpenDark.

obelisk79, 2024-02-02:
> @obelisk79 I am seeing the same issues he reported... I am on 0.21.2 for Mac Intel. The problem goes away with OpenDark.

Thanks for the update. I think we are only a couple of months out from the next release where this is supposedly already fixed. I'm glad to hear that OpenDark addresses the issue though. I'm quite close to an initial release of OpenLight, so if dark themes aren't your preferred style there should be another option.