FreeCAD/FreeCAD #9920: Exporting to an STL after merging files, results in a partial (sometimes) or overlaid (other times) STL file, if the merged file has to be moved (on the X/Y/Z planes). The object looks ok i FreeCAD but the STL file is not as expected.

nhc155, 2023-07-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

When I merged and then moved the new object to where I needed it, the exported STL had the new object in its original position.

With the one I have attached pictures of, the STL file is missing part of the merged file.
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nhc155, 2023-07-14:
Also, please note that the object was mirrored.
I have found that I can better do this in the PART workbench and doing the mirror there, then exporting. This worked. Not sure if this bug report should remain open ?

grubertechnic, 2023-07-14:
Pretty sure, that the wrong object was selected for the export, see [Most Commonly Asked Questions](https://forum.freecad.org/viewtopic.php?p=170906#p170906) point 1.

If this is the case, please consider to close it.

Else please add a file and give a step by step instruction to replicate the issue.

nhc155, 2023-07-14:
Thank you - I was using Box element selection (Shift + E) which works on simple models and did not know I could simply select the body (or both bodies in this case).

closed 2023-07-14 by nhc155