FreeCAD/FreeCAD #9923: [Problem]  wrong rendered in 0.20.2 and 0.21.0 weekly, but not in 0.19

Bastlwastl84, 2023-07-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.60n
  * Alternate_OpenSCAD 1.0.0
  * Assembly4 0.50.2
  * BOLTSFC
  * CubeMenu
  * CurvedShapes 1.0.4
  * Curves 0.6.12
  * Defeaturing 1.2.0
  * Design456 0.0.1
  * dodo 1.0.0
  * dxf-library
  * fasteners 0.4.56
  * frame 0.1.0
  * freecad.gears 1.0.0
  * ImportNURBS 1.1.0
  * LCInterlocking
  * Lithophane
  * MakerWorkbench 1.0.1
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * MOOC 2022.4.21
  * nurbs
  * OSE3dPrinter
  * Part-o-magic 1.1.0
  * parts_library
  * parts_library.backup1689257507.8666701 (Disabled)
  * Plot 2022.4.17
  * Pyramids-and-Polyhedrons
  * retr3d
  * Silk 0.1.3
  * slic3r-tools
  * symbols_library
  * ThreadProfile 1.87.0
  * toSketch 1.0.1
  * workfeature
  * yaml-workspace 0.1.2
[/code]

Version 0.19:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
[... 218 lines cut ...]
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'osecore'
17:38:13  --------------------------------------------------------------------------------
17:38:15  DEBUG:matplotlib:(private) matplotlib data path: C:\Program Files\FreeCAD 0.19\bin\lib\site-packages\matplotlib\mpl-data
17:38:15  DEBUG:matplotlib:matplotlib data path: C:\Program Files\FreeCAD 0.19\bin\lib\site-packages\matplotlib\mpl-data
17:38:15  DEBUG:matplotlib:CONFIGDIR=C:\Users\Unbekannt\.matplotlib
17:38:15  DEBUG:matplotlib:matplotlib version 3.3.3
17:38:15  DEBUG:matplotlib:interactive is False
17:38:15  DEBUG:matplotlib:platform is win32
17:38:15  DEBUG:matplotlib:loaded modules: ['sys', 'builtins', '_frozen_importlib', '_imp', '_warnings', '_frozen_importlib_external', '_io', 'marshal', 'nt', '_thread', '_weakref', 'winreg', 'time', 'zipimport', '_codecs', 'codecs', 'encodings.aliases', 'encodings', 'encodings.utf_8', 'encodings.cp1252', '_signal', '__main__', 'encodings.latin_1', '_abc', 'abc', 'io', '_stat', 'stat', '_collections_abc', 'genericpath', 'ntpath', 'os.path', 'os', '_sitebuiltins', '_locale', '_bootlocale', 'types', 'importlib._bootstrap', 'importlib._bootstrap_external', 'warnings', 'importlib', 'importlib.machinery', 'importlib.abc', '_operator', 'operator', 'keyword', '_heapq', 'heapq', 'itertools', 'reprlib', '_collections', 'collections', '_functools', 'functools', 'contextlib', 'importlib.util', 'mpl_toolkits', 'site', 'FreeCAD', '__FreeCADBase__', 'enum', '_sre', 'sre_constants', 'sre_parse', 'sre_compile', 'copyreg', 're', 'token', 'tokenize', 'linecache', 'traceback', '_opcode', 'opcode', 'dis', 'collections.abc', 'inspect', 'math', '_datetime', 'datetime', 'femtools', 'femtools.migrate_app', 'Asm4_locator', 'utils', 'utils.preferences', '_weakrefset', 'weakref', 'pkgutil', 'freecad', 'freecad.toSketch', 'freecad.ose3dprinter', 'freecad.importNURBS', 'pygears', 'freecad.gears', 'freecad.frametools.version', 'freecad.frametools', 'freecad.Curves.version', 'freecad.Curves', '__future__', 'binascii', 'posixpath', 'fnmatch', 'errno', 'zlib', '_compression', 'threading', '_bz2', 'bz2', '_lzma', 'lzma', 'shutil', '_struct', 'struct', 'zipfile', 'platform', 'xml', 'xml.parsers', 'pyexpat.errors', 'pyexpat.model', 'pyexpat', 'xml.parsers.expat.model', 'xml.parsers.expat.errors', 'xml.parsers.expat', 'plistlib', 'email', 'email.errors', '_string', 'string', 'email.quoprimime', 'base64', 'email.base64mime', 'quopri', 'email.encoders', 'email.charset', 'email.header', '_bisect', 'bisect', '_sha512', '_random', 'random', '_socket', 'select', 'selectors', 'socket', 'urllib', 'urllib.parse', 'locale', 'calendar', 'email._parseaddr', 'email.utils', 'email._policybase', 'email.feedparser', 'email.parser', 'tempfile', 'textwrap', 'pkg_resources.extern', 'pkg_resources._vendor', 'pkg_resources._vendor.six', 'pkg_resources.extern.six', 'pkg_resources._vendor.six.moves', 'pkg_resources.extern.six.moves', '_ctypes', 'ctypes._endian', 'ctypes', 'pkg_resources._vendor.appdirs', 'pkg_resources.extern.appdirs', 'pkg_resources._vendor.packaging.__about__', 'pkg_resources._vendor.packaging', 'pkg_resources.extern.packaging', 'pkg_resources.extern.packaging._structures', 'pkg_resources.extern.packaging.version', 'pkg_resources.extern.packaging._compat', 'pkg_resources.extern.packaging.specifiers', 'copy', 'pprint', 'pkg_resources._vendor.pyparsing', 'pkg_resources.extern.pyparsing', 'pkg_resources._vendor.six.moves.urllib', 'pkg_resources.extern.six.moves.urllib', 'pkg_resources.extern.packaging.markers', 'pkg_resources.extern.packaging.requirements', 'sysconfig', 'pkg_resources', 'freecad.OpenSCAD_Alt_Import', 'freecad.cadenas3dfindit', 'freecad.plot.version', 'freecad.plot', 'FreeCADGui', 'codeop', 'code', 'pivy.pivy_meta', 'swig_runtime_data4', 'pivy._coin', 'pivy.coin', 'pivy', 'typing.io', 'typing.re', 'typing', 'signature_bootstrap', 'shiboken2.shiboken2', 'shibokensupport', 'shibokensupport.signature', 'shibokensupport.__feature__', 'shibokensupport.signature.typing', 'shibokensupport.signature.inspect', 'shibokensupport.signature.lib', 'shibokensupport.signature.lib.tool', 'shibokensupport.signature.mapping', 'shibokensupport.signature.errorhandler', 'shibokensupport.signature.layout', 'shibokensupport.signature.parser', 'shibokensupport.signature.importhandler', 'shibokensupport.signature.lib.enum_sig', 'PySide2.support.__feature__', 'PySide2.support.signature', 'PySide2.support.signature.mapping', 'PySide2.support.signature.errorhandler', 'PySide2.support.signature.layout', 'PySide2.support.signature.lib', 'PySide2.support.signature.parser', 'PySide2.support.signature.importhandler'...
17:38:15  DEBUG:matplotlib:CACHEDIR=C:\Users\Unbekannt\.matplotlib
17:38:15  DEBUG:matplotlib.font_manager:Using fontManager instance from C:\Users\Unbekannt\.matplotlib\fontlist-v330.json
17:38:15  DEBUG:matplotlib.pyplot:Loaded backend qt5agg version unknown.
17:38:15  DEBUG:matplotlib.pyplot:Loaded backend Qt5Agg version unknown.
17:38:15  DEBUG:matplotlib.pyplot:Loaded backend module://freecad.plot.freecad_backend version unknown.

### Anything else?

from here https://github.com/ogland/Printer-mods/tree/main/V2/CPAP

this file : https://github.com/ogland/Printer-mods/blob/main/V2/CPAP/CAD/SB-CPAP.step

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Bastlwastl84, 2023-07-14:
i've marked the shown "Faces" (green)

grubertechnic, 2023-07-14:
Thank you for the many details, but even you have marked where the issue is, what is the issue? At least i see none when i import it. That many parts overlap is hopefully obvious.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33581 (Git)
Build type: Release
Branch: master
Hash: 0c185ad6117f37da5cb467416750318196611430
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

bdieterm, 2023-07-16:
I can kind of reproduce the error, and "switch it on/off" within the same FreeCAD instance:
start FreeCAD normally
switch to the Part workbench
open the Preferences dialog
Part/Part Design -> Shape view
set "Maximum angular deflection" to 28.5° and "Maximum deviation" to 0.5 %
press OK
load the step file
--> no error in the geometry
open the Preferences dialog again 
set "Maximum deviation" to 0.01 %
press Apply
--> the geometry is instantly updated and the holes are damaged
(changing "Maximum deviation" back to a higher value does not undo the damage)

So, the error occurs when, during step import, either the "Maximum deviation" value is too low or the "Maximum angular deflection" is too low.

@Bastlwastl84: Which values do you use? Can you try higher values?
@grubertechnic: Can you try lower values?

For me, the threshold was a deviation of 0.017 % (good) and 0.016 % (bad), with "Maximum angular deflection" of 28.5°.

grubertechnic, 2023-07-22:
Yes, with finer deviations the issue occurs. A few notes:

 - A deviation of 0.01% is insane, it just consumes calculation time.
 - When changing the tessellation in the preferences and apply to it, should not change already used and created objects in the tree, only new imported loaded or new created (explicit new). That seems to be a bug.
 - A head of OCCT7.7.2 is created a few hours ago, see https://git.dev.opencascade.org/gitweb/?p=occt.git ; there is also a tessellation bug from steps in the should be fixing list of 7.7.2 (maybe not merged already, see https://tracker.dev.opencascade.org/roadmap_page.php). Maybe the bug is solved in a certain time (or maybe not).

bdieterm, 2023-07-23:
Hi @grubertechnic!

>     * A deviation of 0.01% is insane, it just consumes calculation time.
... and also memory (RAM). I totally agree. Looking at Bastlwastl's screenshots I have the impression that lower values than the default values were used because the circles look smoother than on my system.
 
>     * When changing the tessellation in the preferences and apply to it, should not change already used and created objects in the tree, only new imported loaded or new created (explicit new). That seems to be a bug.
I had a deeper look at this. The objects (shapes) are not modified when changing the "deviation" or "angular deflection" parameters. Only the temporary mesh that is used for displaying the geometries is affected. The missing update of the view mesh when going back to a coarser resolution seems to be a bug indeed. I just created a fix for that: #9972

grubertechnic, 2023-07-23:
> had a deeper look at this. The objects (shapes) are not modified when changing the "deviation" or "angular deflection" parameters. Only the temporary mesh that is used for displaying the geometries is affected.

For me, also the objects itself are changed.

[2023-07-23 12-25.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/ff528c82-1724-4211-9039-e59a0443754d)

As far as i know no `Object Style` Property should change when apply for good reasons. For example i have one very complicated object, which i want to have rougher to save time, it overwrites it. Same with the color, but here it works as expected.

bdieterm, 2023-07-23:
@grubertechnic 
You are right. It is even worse. Just opening the Preferences dialog and clicking OK or Apply will modify the values in all objects in all open files. As far as I have seen there is no other use for these properties other than for initializing a new object. This is a bug.

grubertechnic, 2023-07-23:
Note: i am not 100% sure, if that it should work in that way, but as i wrote, this is as i like remember. Maybe i am wrong. Before waking up sleeping dogs, i will search tomorrow in the forum, what was the last state of the art.

grubertechnic, 2023-07-24:
@0penBrain : i need a decision here, if that is a bug or not, to make a ticket or not:

When changing in the preferences the tessellation and press apply, should it change already tessellated objects? As far is i know it should not. Also here in the forum, https://forum.freecad.org/viewtopic.php?p=451634#p451634 (post from you), it is also written here, but not explicit when _Apply_. What do you think about?

Due "not good search tool" and also using a search engine, i do not find a better answer. In worst case i can open a ticket/thread, but often when on pining, i do not get (m)any answers.

Thanks in advance.

Bastlwastl84, 2023-07-25:
It also happens if creating a mesh and exporting. Than the bug is also
visible in windows 3d viewer if the details are high.

adrianinsaval, 2023-07-25:
> When changing in the preferences the tessellation and press apply, should it change already tessellated objects?

I think that is intended behavior.

grubertechnic, 2023-07-25:
> I think that is intended behavior.

Note, this is also when clicking OK. That was definitely not always intended.

Generally speaking, it makes in that case more sense like in the sketcher, to add an extra button for that, _Apply to existing models_ (or objects).

[image]

0penBrain, 2023-07-27:
> @0penBrain : i need a decision here, if that is a bug or not, to make a ticket or not:
>
> When changing in the preferences the tessellation and press apply, should it change already tessellated [...]

Not sure, but I think in the past it didn't change the already created objects.
Which I think is good.
Or at least forcefully applying to existing objects (BTW, does that touch only active doc, or all open docs) can ruin one's effort that would have carefully set different values for lot of objects in his/her current project. :)
Also because of how preferences generally works, I'm afraid that even if user doesn't change this values, they still will be applied to already created objects (didn't check that though).
So yes, IMO it's a bug.
My proposal:
 * Short fix => Do not touch already created objects
 * Long fix => If values changed, ask the user if already created objects should be impacted or not
Please open a ticket (you can ping me there) so this thread can focus back on the original issue. :wink:

grubertechnic, 2023-07-27:
> Not sure, but I think in the past it didn't change the already created objects.
> Which I think is good.
> Or at least forcefully applying to existing objects (BTW, does that touch only active doc, or [...]

This is also what i think.

> Please open a ticket (you can ping me there) so this thread can focus back on the original issue.

Yes, i will do.

luzpaz, 2023-10-29:
What's the status of this ticket ?

maxwxyz, 2024-08-13:
@Bastlwastl84 is this still an issue in the latest dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

MiqCG, 2024-08-26:
Hi,

If it helps, I can still reproduce the issue with the steps provided by @bdieterm in revision 38459.

1. start FreeCAD 
2. switch to the Part Desigh workbench
3. open the Preferences > Part Design -> Shape view
4. set "Maximum angular deflection" to 28.5° and "Maximum deviation" to 0.5 %
5. press OK
6. Import the step file with default options --> no error in the Report view, and no errors on Recompute
7. open the Preferences dialogue again
9. set "Maximum deviation" to 0.01 %
10. press OK
11. problems appear on the body at least in the bottom holes.

[image]

You can find the files here (5,8 and 13,2 Mb): 
https://mega.nz/folder/astEyCjZ#Fnq5mzRGC-2FO5MWx_ZfkQ

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git) AppImage
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * sheetmetal 0.4.24
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * fasteners 0.5.27

luzpaz, 2024-12-09:
@MiqCG Hi would you mind testing this on latest 1.1-dev  ?

MiqCG, 2024-12-10:
> Can you please paste full about info of revision you tested on? TIA

Of course, although it was already indicated in my previous message, below the two images.

```
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

Syres916, 2025-10-25:
Just as an aside, I find to balance performance and reasonable rendering quality, I use the following:

[image]

stefano2734, 2026-09-22:
Should be tested with newer occt 8.0.1 in 26.3, which will come for all systems with 8.0.1 in next days.
Perhaps improved or solved with some solved issues after occt 7.8.1 in relation to step.

flaviut, 2026-11-16:
I can reproduce this issue on both OCCT 7.9.3 and 8.0.1, it doesn't seem to have been resolved upstream.

https://github.com/Open-Cascade-SAS/OCCT/issues/572 seems to have the same underlying root cause.