FreeCAD/FreeCAD #9924: [Feature Request] Adding an arc length constrain function

NXTler, 2023-07-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git) AppImage
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * Curves 0.6.12
  * CfdOF 1.23.1
  * TabBar
  * IconThemes
  * Glass
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Hello all,
after some casual part making, I have discovered that FreeCAD is currently lacking an arc length constrain function.
Arc length constrains are mostly used for gears and other complex geometries, hence most CAD programs have it implemented in some form.
If you want to create an arc length constrain in FreeCAD, you are forced to use the Spreadsheet. Which, in my opinion, adds unnecessary complexity and feels disruptive to the overall workflow.

I would love to hear other opinions on this topic.
Many thanks in advance!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-07-15:
Forum topic with a workaround:
https://forum.freecad.org/viewtopic.php?t=79443

grubertechnic, 2023-07-15:
> Forum topic with a workaround

Maybe there is something lost in translation on my side, but in the from that is the chord length, while this here is the arc length.

As workaround: in general the arc length you do not need any spreadsheet. You can workaround this with expressions, just add two lines on the end of the arc to the circle center and add the arc length in the expressions of the angle.

NXTler, 2023-07-15:
That's correct, but I still think it's unnecessarily complex just for setting the arc length.

grubertechnic, 2023-07-15:
I do not say, that your feature request is invalid. I just wanted to note, that for the workaround no spreadsheet is needed.

NXTler, 2023-07-15:
Sorry, I didn't want to sound mean. English is not my native language.

grubertechnic, 2023-07-22:
> Sorry, I didn't want to sound mean.

I see nothing mean in your comment. Quite the contrary, i was maybe not explicit enough.

> English is not my native language.

+1

Zbuser93, 2024-02-24:
Unfortunately none of the workarounds work in all cases. You can currently only constrain arc length if the radius or the internal angle of the arc is known, but if both of those are unknown and only the arc length is known, then there is currently no way to constrain this. I just wanted to add this to the conversation to show that this feature request really is substantial, and not just a matter of convenience.

PR #12602 (2024-04-01): Sketcher : ArcLength Constraint
Extend constraint distance to arc length.
Fix #9924 

Current status:
- [x] App part
[image]

- [x] Gui part
closed 2024-04-01 by sliptonic