FreeCAD/FreeCAD #9926: [Feature Request] Dropdown menus for custom Toolbars

NXTler, 2023-07-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git) AppImage
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * Curves 0.6.12
  * CfdOF 1.23.1
  * TabBar
  * IconThemes
  * Glass
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Core

### Problem description

Hello all,
every CAD user have their own preferences in regard to their application. So customization is key for a good workflow and experience.
With custom Toolbars you are already able to achieve a decent amount of customization, though dropdown menus can not be created like it is seen with the default Toolbars. I think dropdown menus can be quite handy to bundle up similar functions for a more space efficient usage and should be easy to create.

I would love to hear other opinions on this topic.
Many thanks in advance!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-07-15:
Can you illustrate or mockup what your are referring to ?

NXTler, 2023-07-15:
These are the dropdown menus by the default toolbars:
[image]
My request is about being able to create custom dropdown menus using custom toolbar:
[image]

This is a screenshot of Solid Edges customization with the dropdown menu button and the corresponding result highlighted as an example (Sorry, it's the best one I have found):
[image]

luzpaz, 2023-07-15:
Yea, that makes more sense now. Thanks for the elaboration.