FreeCAD/FreeCAD #9927: Incorrect generation of indexes in windows array (parametric count)

kaiwas, 2023-07-15:
Hi!
An array of windows that counts the number of array elements according to some rules (wall length) does not correctly assign indexes to array elements.
The error occurs both when the wall length increases and when the wall length decreases. But a certain logic could not be detected.

Temporary solution to the problem.  To assign in the array settings Expand Array - > false.
Save the file. Close it and open it.
Then do Expand Array - > True.

Video example 
https://youtu.be/PbrzLWttXDM

File https://disk.yandex.ru/d/vKtR90BAvnkA1w

-----
FreeCAD_weekly-builds-33458-2023-07-02-conda-Linux-x86_64-py310.AppImage
Linux Manjaro

luzpaz, 2023-07-15:
CC @yorikvanhavre

Roy-043, 2023-07-15:
A new issue has been created to handle the array count problem:
https://github.com/FreeCAD/FreeCAD/issues/9912
PR:
https://github.com/FreeCAD/FreeCAD/pull/9929

Therefore closing this.

closed 2023-07-15 by Roy-043