FreeCAD/FreeCAD #9933: [Problem]

jamwaffles, 2023-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33576 (Git)
Build type: Release
Branch: master
Hash: 8e6fbd2886d3ad51b75232bd0019d92d496e80cb
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Path

### Problem description

I'm trying to add a `LeadInOut` dressup to a `Profile` operation in the Path workbench, however I get this exception:

```
11:01:04  Traceback (most recent call last):
  File "C:\Users\jamwa\Downloads\FreeCAD_weekly-builds-33576-2023-07-13-conda-Windows-x86_64-py310\Mod\Path\Path\Dressup\Gui\LeadInOut.py", line 181, in execute
    obj.Path = self.generateLeadInOutCurve(obj)
  File "C:\Users\jamwa\Downloads\FreeCAD_weekly-builds-33576-2023-07-13-conda-Windows-x86_64-py310\Mod\Path\Path\Dressup\Gui\LeadInOut.py", line 577, in generateLeadInOutCurve
    temp = self.getLeadStart(obj, layer, action)
  File "C:\Users\jamwa\Downloads\FreeCAD_weekly-builds-33576-2023-07-13-conda-Windows-x86_64-py310\Mod\Path\Path\Dressup\Gui\LeadInOut.py", line 285, in getLeadStart
    offsetvector = self.multiply(offsetvector, -1)
<class 'AttributeError'>: 'ObjectDressup' object has no attribute 'multiply'
11:01:04  Recompute failed!
```

Steps to reproduce:

1. Open the attached file in the FreeCAD version listed (`FreeCAD_weekly-builds-33576-2023-07-13-conda-Windows-x86_64-py310` from the weekly builds)
2. Go to Path workbench
3. Click on the `Profile` operation (last in the op list)
4. Go to <kbd>Path</kbd> -> <kbd>Path Dressup</kbd> -> <kbd>LeadInOut</kbd>
5. Don't change any settings (defaults for me are type `Arc`, 6mm radius, etc
6. See that the console prints the above traceback.

### Anything else?

[gas-spring-clamp.zip](https://github.com/FreeCAD/FreeCAD/files/12061095/gas-spring-clamp.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jamwaffles, 2023-07-15:
I forgot to add a title. It's there now - sorry!

rmu75, 2023-07-15:
I can reproduce and I think I have a fix.

jamwaffles, 2023-07-15:
To prevent duplicate work, I think I already fixed it in #9935 but I could've missed something.

rmu75, 2023-07-15:
Exactly the same thing I have done :-). I think it would not hurt adding a testcase.

jamwaffles, 2023-07-15:
You're welcome to push to that PR or close it in favour of your own. I don't write Python so I wouldn't know where to begin adding a test, although I do agree it could use one. I'll let you take it from here :)

closed 2023-07-18 by sliptonic