FreeCAD/FreeCAD #9936: [Problem] Can't add LeadInOut dressup to Adaptive operation in Path WB

jamwaffles, 2023-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33576 (Git)
Build type: Release
Branch: master
Hash: 8e6fbd2886d3ad51b75232bd0019d92d496e80cb
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Problem description

Trying to add a `LeadInOut` dressup to an adaptive operation in Path but I get this error:

```
11:23:29  Traceback (most recent call last):
  File "C:\Users\jamwa\Downloads\FreeCAD_weekly-builds-33576-2023-07-13-conda-Windows-x86_64-py310\Mod\Path\Path\Dressup\Gui\LeadInOut.py", line 181, in execute
    obj.Path = self.generateLeadInOutCurve(obj)
  File "C:\Users\jamwa\Downloads\FreeCAD_weekly-builds-33576-2023-07-13-conda-Windows-x86_64-py310\Mod\Path\Path\Dressup\Gui\LeadInOut.py", line 549, in generateLeadInOutCurve
    and curCommand.z < currLocation["Z"]
<class 'TypeError'>: '<' not supported between instances of 'NoneType' and 'float'
11:23:29  Recompute failed!
```

Steps to reproduce:

1. Open the attached file in the FreeCAD version listed (1FreeCAD_weekly-builds-33576-2023-07-13-conda-Windows-x86_64-py3101 from the weekly builds)
2. Go to Path workbench
3. Click on the operation named `Adaptive` and enable it with <kbd>Space</kbd>
4. Go to <kbd>Path</kbd> -> <kbd>Path Dressup</kbd> -> <kbd>LeadInOut</kbd>
5. Don't change any settings (defaults for me are type `Arc`, 6mm radius, etc
6. See that the console prints the above traceback.

I added an `is not None` check to the line mentioned, but then got another `cannot normalised a null vector` error from another place in that file, so I gave up trying to fix it as I don't know Python enough.

### Anything else?

[gas-spring-clamp.zip](https://github.com/FreeCAD/FreeCAD/files/12061142/gas-spring-clamp.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-07-15:
This seems to touch @sliptonic's code.

luzpaz, 2024-08-30:
@firbladed is this issue reproducible for you ? (CC @jamwaffles)

firebladed, 2024-08-31:
Initially i just said as possibly related because was same dress up operation with  same error

just testing on dev version its doesn't get that error for me but its behavior is weird, currently it seams to generate a lead in and lead out on every arc generated by adaptive operation which i think is wrong

on the provided gas spring clamp example

First i get no errors so that appears fixed.
however i also get the multiple lead in and lead outs problem with one for every arc  generated by adaptive rather than just a pair for the whole adaptive operation which i assume is whats supposed to happen

so id say this is technically fixed, but still has other issues

closed 2025-02-06 by github-actions[bot]