FreeCAD/FreeCAD #9938: [Problem] Copy and paste not working on spreadsheet

jorgerobles15, 2023-07-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

when I try to use copy and paste (either using CTRL+C or via edit menu) the program gives a "(Not Responding)" message and stays like that for more than 40 secs. This happens even when I try to copy one single cell. 
thanks!

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-07-15:
Many times I copied and never I encountered this. I've just tried to reproduce the issue. Still, this freeze does not occur. I think, the problem comes from a Windows 10 update. Maybe the next one will solve the problem. Or the next after the next.

```ruby
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

jorgerobles15, 2023-07-15:
thanks a lot, this is my Windows info:
Edition Windows 11 Home
Version 22H2
Installed on ‎2023-‎03-‎18
OS build 22621.1992
Experience Windows Feature Experience Pack 1000.22644.1000.0

thanks

luzpaz, 2023-07-16:
Please try to reproduce on development version https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

xtemp09, 2023-07-16:
This is definitely a bug ― FreeCAD reports incorrect OS version. I would say that this issue is urgent.

chennes, 2023-07-16:
I believe we fixed (or really, mitigated) this already in 0.21 -- it had to do with the Windows Accessibility stuff. Please confirm.

closed 2023-10-05 by Roy-043