FreeCAD/FreeCAD #9940: No option in Job/2d I have to select every lin

cgerrist, 2023-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
0.20.2
```

### Subproject(s) affected?

Path

### Problem description

I am busy with a drawing with more the 1000 separate lines. When I want to plot the drawing via freecad I have to select every line whenI create a Job. A new option ‘All’ will help a lot.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

cgerrist, 2023-07-16:
Select all exist

cgerrist, 2023-07-16:
I found the option

closed 2023-07-16 by cgerrist

luzpaz, 2023-07-16:
@cgerrist JFYI, we aren't backporting fixes to v0.20.x  
If you want to file a ticket, we need it to be a bug from v0.21.x (development build) which can be downloaded from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds. We also need full [About](https://wiki.freecad.org/About) info with every ticket (this saves us a lot of time and manpower). TIA!