FreeCAD/FreeCAD #9941: wrong tooltip text for "Import invisible objects" in STEP import settings

bdieterm, 2023-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
-
```

### Subproject(s) affected?

None

### Problem description

There is a contradiction between the label and the tooltip text

see
https://github.com/FreeCAD/FreeCAD/blob/15371a04f38bbeb5a04edc5a91020ada1e30ef88/src/Mod/Part/Gui/DlgImportStep.ui#L58-L63

The tooltip text is wrong (and not the label text).
Fixing this would also require updated translations.

### Anything else?

There is a contradiction between the label and the tooltip text in the STEP file import settings.

see
https://github.com/FreeCAD/FreeCAD/blob/15371a04f38bbeb5a04edc5a91020ada1e30ef88/src/Mod/Part/Gui/DlgImportStep.ui#L58-L63

The tooltip text is wrong (and not the label text).
Fixing this would also require updated translations.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-07-16:
Thanks!  
Question for anyone: What is our current definition of 'invisible objects' ? That also would be useful to add to the tooltip, imho.

benj5378, 2023-07-17:
@luzpaz Perhaps invisible objects are just toggled objects in the tree view? https://wiki.freecad.org/Std_ToggleVisibility

luzpaz, 2023-07-18:
@benj5378 thanks for spearheading it. Closing as fixed by you.

closed 2023-07-18 by luzpaz

benj5378, 2023-07-18:
@luzpaz plz check forum