FreeCAD/FreeCAD #9942: Select all in Job not working

cgerrist, 2023-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 13.4
Word size of FreeCAD: 64-bit
Version: 0.21.0.31483 (Git)
Build type: Release
Branch: master
Hash: 588620fd9fd138892b1cd2466f5c87d441493174
Python 3.11.0, Qt 5.15.6, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0
  * symbols_library
  * ModernUI
  * sheetmetal 0.2.63
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Path

### Problem description

I have a SVG with a lot of not connected lines I want to select them all.
I will use this for my cnc/plotter,

the Select All in Job/2d right click available  but I cannot select it.

Is is in the Path/Job/2d when you see more lines the first is selected. 
Do a right click on the 2d and you get a little menu with as last the 'Select All' option.
If I have a few lines I can select it manually but I have a plot with more the 1000 separate lines.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

cgerrist, 2023-07-18:
> @cgerrist <https://github.com/cgerrist> can you test on latest v0.21 to make sure ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds (you're using a build that's a few hundred commits shy of latest dev build)

I updated the problem. But it still doesn't work. Select all is disabled.

￼

luzpaz, 2023-07-18:
@cgerrist can you post the [About](https://wiki.freecad.org/About) info of the latest dev version you tested ? TIA!

cgerrist, 2023-07-18:
Done

cgerrist, 2023-07-18:
> 
> Done
> 
>> 
>> 
>> @cgerrist <https://github.com/cgerrist> can you test on latest v0.21 to make sure ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds (you're using a build that's a few hundred commits shy of latest dev build)

I did another update

luzpaz, 2023-09-10:
@cgerrist can you upload example SVG for testing ?

luzpaz, 2023-09-23:
@cgerrist :point_up_2:

luzpaz, 2023-12-17:
@cgerrist ping

luzpaz, 2024-12-13:
@cgerrist ping

luzpaz, 2025-05-10:
@cgerrist can you reproduce this on 1.0 or 1.1dev ?

sliptonic, 2025-06-02:
Close it.  Happy to re-open if he returns

closed 2025-08-17 by github-actions[bot]