FreeCAD/FreeCAD #9947: [Problem] Crash in preferences when clicking workbenches after having clicked help button

benj5378, 2023-07-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33410 (Git)
Build type: Debug
Branch: iss5 (branched from 32d8029780b23b176433d149a10e47e42f00274b)
Hash: ab8b64fa2bd07d65322fa1cf20f9256e668f10af
Python 3.10.7, Qt 5.15.6, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Problem description

When the help button in the preferences window is pressed followed by a click on Start preferences, FreeCAD crashes.

How to reproduce
1) Start FreeCAD
2) Open preferences: Edit > Preferences
3) Click the help button. (don't click anywhere else until next step)
4) Click on the button for the Start preferences
5) Crash

The crash only happens when clicking the buttons sometimes. Some people have easier reproducing it, though it is reproducable across users.

https://forum.freecad.org/viewtopic.php?p=691311#p691311

### Anything else?

Backtrace:
```
malloc(): mismatching next->prev_size (unsorted)

Thread 2 "QXcbEventQueue" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffec08e6c0 (LWP 8093)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt full
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:44
        tid = <optimized out>
        ret = 0
        pd = <optimized out>
        old_mask = {__val = {93824994487248}}
        ret = <optimized out>
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>)
    at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6)
    at ./nptl/pthread_kill.c:89
#3  0x00007ffff163bc46 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/posix/raise.c:26
        ret = <optimized out>
#4  0x00007ffff16227fc in __GI_abort () at ./stdlib/abort.c:79
        save_stage = 1

                  act = {__sigaction_handler = {sa_handler = 0x20, sa_sigaction = 0x20}, sa_mask = {__val = {140737353651504, 0, 140737488341120, 140737145004032, 140737353973969, 5, 93824994487248, 1100, 140737159877216, 93824994375224, 93824998785584, 140737145004032, 140737353973969, 5, 0, 140737251430144}}, sa_flags = -234108048, sa_restorer = 0x55555575f638}
#5  0x00007ffff16850be in __libc_message
--Type <RET> for more, q to quit, c to continue without paging--
   entry=0x7ffff17b95dc "%s\n") at ../sysdeps/posix/libc_fatal.c:155
        ap = {{gp_offset = 24, fp_offset = 0, overflow_arg_area = 0x7fffec08d850, reg_save_area = 0x7fffec08d7e0}}
        fd = <optimized out>
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
#6  0x00007ffff169c3fc in malloc_printerr (str=str@entry=0x7ffff17bc758 "malloc(): mismatching next->prev_size (unsorted)") at ./malloc/malloc.c:5660
#7  0x00007ffff169f9bc in _int_malloc (av=av@entry=0x7fffe4000030, bytes=bytes@entry=124) at ./malloc/malloc.c:4003
        next = <optimized out>
        iters = <optimized out>
        nb = <optimized out>
        idx = <optimized out>
        bin = <optimized out>
        victim = <optimized out>
        size = <optimized out>
        victim_index = <optimized out>
        remainder = <optimized out>
        remainder_size = <optimized out>
        block = <optimized out>
        bit = <optimized out>
        map = <optimized out>
        fwd = <optimized out>
        bck = <optimized out>
        tcache_unsorted_count = 1
        tcache_nb = <optimized out>
        tc_idx = 7
        return_cached = <optimized out>
        __PRETTY_FUNCTION__ = "_int_malloc"
#8  0x00007ffff16a06fd in __GI___libc_malloc (bytes=124) at ./malloc/malloc.c:3323
        ar_ptr = 0x7fffe4000030
        victim = <optimized out>
        tbytes = <optimized out>
        __PRETTY_FUNCTION__ = "__libc_malloc"
#9  0x00007ffff078e1e2 in  () at /lib/x86_64-linux-gnu/libxcb.so.1
#10 0x00007ffff078ea0e in  () at /lib/x86_64-linux-gnu/libxcb.so.1
#11 0x00007ffff079028c in xcb_wait_for_event () at /lib/x86_64-linux-gnu/libxcb.so.1
#12 0x00007fffec5ba788 in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#13 0x00007ffff1eccc51 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff1690402 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
        ret = <optimized out>
        pd = <optimized out>

                      unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737153394368, 2572706676385403850, -400, 11, 140737488341120, 140737145004032, -2572735133904120886, -2572692533934009398}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#15 0x00007ffff171f590 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2023-07-17:
https://forum.freecad.org/viewtopic.php?p=691311#p691311

benj5378, 2023-07-17:
@xtemp09

> Can I ask, what is Branch: iss5 (branched from 32d8029780b23b176433d149a10e47e42f00274b)?

Just a branch that I was working on, but that shouldn't be of any concern. 

Full version info shows that I am using a Debug version. What do you want to know about the compiler and debugger? It's just gcc 12.2.0 and gdb. Build was configured with standard options. Yes, I can reproduce it in a release version.

xtemp09, 2023-07-17:
One can foggily see that this is a Qt bug. Qt's bugtracker and github are filled with issues remotely similar to this.

There are two ways:

- Try to catch the function that causes the trouble and trace up to the malloc mismatch. Perhaps, it can be done with _valgrind_.
- Compile FreeCAD against a debug version of Qt

I would try to do this myself, but I can't reproduce the crash. ='(

Anyway, this bug does not come from the source code of FreeCAD. If you find the culprit, the developers of FreeCAD might be able to apply a workaround.

benj5378, 2023-07-17:
It's also an edge case that and really rarely happens. Would just be sad, if one had a big model open with unsaved changes and this happens.

xtemp09, 2023-07-17:
@benj5378, it seems to be a Qt bug, an XCB bug or a driver problem. Workaround can be applied by somebody who can reproduce this crash. It means, you or openBrain have to compile FreeCAD against the debug version of Qt or back trace it using additional debugging techniques.

If no developer/contributor is able to reproduce it, the issue will be open indefinitely.

P.S. You and openBrain should write your computer configuration. I bet $1 you have Nvidia ― Google showed bug reports similar to yours.

benj5378, 2023-07-17:
@xtemp09 even if we got the trace, it's probably not going to be a quick fix. I agree that it might be Qt, that was also what my own investigation at the time suggested.

If I went around and betted on the street if one had an Nvidia graphics card, I would always be in a surplus, because the vast majority has Nvidia (if not talking about interegrated graphics), so it's not saying much, if just me and OpenBrain happen to use Nvidia. Though I could test it on Intel integrated graphics.

luzpaz, 2023-08-21:
Any relationship to #10176

benj5378, 2023-08-21:
> Any relationship to #10176 ?

@luzpaz No, this is a C++ error causing segfault with tracebacks leading to Qt. The issue you are mentioning is a Python error caused by some naming error.

xtemp09, 2024-02-12:
I compiled a debug version of Qt and finally reproduced the crash. One can see that this is a purely Qt bug. I've searched the bug tracker, there are many related bugs, some of them have been fixed in Qt 6, some of them are still there.

Here is the backtrace.

<details><summary>The backtrace</summary>

```
malloc(): mismatching next->prev_size (unsorted)
Process 34983 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
    frame #0: 0x00007ffff34969fc libc.so.6`__GI___pthread_kill at pthread_kill.c:44:76
```

```cpp
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff34969fc libc.so.6`__GI___pthread_kill at pthread_kill.c:44:76
    frame #1: 0x00007ffff34969b0 libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_internal(signo=6, threadid=140737259076224) at pthread_kill.c:78:10
    frame #2: 0x00007ffff34969b0 libc.so.6`__GI___pthread_kill(threadid=140737259076224, signo=6) at pthread_kill.c:89:10
    frame #3: 0x00007ffff3442476 libc.so.6`__GI_raise(sig=6) at raise.c:26:13
    frame #4: 0x00007ffff34287f3 libc.so.6`__GI_abort at abort.c:79:7
    frame #5: 0x00007ffff3489676 libc.so.6`__libc_message(action=do_abort, fmt="") at libc_fatal.c:155:5
    frame #6: 0x00007ffff34a0cfc libc.so.6`malloc_printerr(str=<unavailable>) at malloc.c:5664:3
    frame #7: 0x00007ffff34a437c libc.so.6`_int_malloc(av=0x00007fffe8000030, bytes=65) at malloc.c:4007:13
    frame #8: 0x00007ffff34a4b8a libc.so.6`_int_realloc(av=0x00007fffe8000030, oldp=0x00007fffe800a500, oldsize=48, nb=80) at malloc.c:4878:20
    frame #9: 0x00007ffff34a5862 libc.so.6`__GI___libc_realloc(oldmem=0x00007fffe800a510, bytes=64) at malloc.c:3494:10
    frame #10: 0x00007ffff3eb223a libQt5Core.so.5`reallocateData(header=0x00007fffe800a510, allocSize=<unavailable>, options=8) at qarraydata.cpp:183:49
    frame #11: 0x00007ffff3eb24fa libQt5Core.so.5`QArrayData::reallocateUnaligned(data=0x00007fffe800a510, objectSize=<unavailable>, capacity=20, options=<unavailable>) at qarraydata.cpp:246:66
    frame #12: 0x00007ffff3f3b311 libQt5Core.so.5`QString::reallocData(unsigned int, bool) [inlined] QTypedArrayData<unsigned short>::reallocateUnaligned(options=<unavailable>, capacity=4, data=<unavailable>) at qarraydata.h:232:78
    frame #13: 0x00007ffff3f3b305 libQt5Core.so.5`QString::reallocData(this=0x00007fffffff91c0, alloc=4, grow=true) at qstring.cpp:2379:44
    frame #14: 0x00007ffff3f3c186 libQt5Core.so.5`QString::append(this=0x00007fffffff91c0, ch=(ucs = 95)) at qstring.cpp:2773:20
    frame #15: 0x00007fffec152e94 libqsvgicon.so`QSvgIconEngine::pixmap(QSize const&, QIcon::Mode, QIcon::State) [inlined] QSvgIconEnginePrivate::pmcKey(state=Off, mode=Normal, size=0x00007fffffff97d8, this=0x00005555588fab30) at qsvgiconengine.cpp:71:57
    frame #16: 0x00007fffec152e39 libqsvgicon.so`QSvgIconEngine::pixmap(this=0x0000555558963320, size=0x00007fffffff97d8, mode=Normal, state=Off) at qsvgiconengine.cpp:199:48
    frame #17: 0x00007ffff44c9baf libQt5Gui.so.5`ScalableEntry::pixmap(this=0x00005555585a8690, size=0x00007fffffff97d8, mode=Normal, state=Off) at qiconloader.cpp:815:48
    frame #18: 0x00007ffff44cfdc6 libQt5Gui.so.5`QIconLoaderEngine::pixmap(this=0x00005555585a8430, size=0x00007fffffff97d8, mode=Normal, state=Off) at qiconloader.cpp:827:47
    frame #19: 0x00007ffff44c6c88 libQt5Gui.so.5`QIcon::pixmap(this=<unavailable>, window=<unavailable>, size=0x00007fffffff97d8, mode=Normal, state=Off) const at qicon.cpp:905:61
    frame #20: 0x00007ffff4d78edc libQt5Widgets.so.5`QCommonStyle::drawControl(this=0x0000555555b59fa0, element=<unavailable>, opt=0x00007fffffff9780, p=0x00007fffffff9ba0, widget=0x00005555585a7d20) const at qcommonstyle.cpp:1389:105
    frame #21: 0x00007ffff4de9788 libQt5Widgets.so.5`QFusionStyle::drawControl(this=0x0000555555b59fa0, element=CE_PushButtonLabel, option=<unavailable>, painter=0x00007fffffff9ba0, widget=0x00005555585a7d20) const at qfusionstyle.cpp:1780:38
    frame #22: 0x00007ffff4de9657 libQt5Widgets.so.5`QFusionStyle::drawControl(this=0x0000555555b59fa0, element=<unavailable>, option=<unavailable>, painter=0x00007fffffff9ba0, widget=0x00005555585a7d20) const at qfusionstyle.cpp:1772:33
    frame #23: 0x00007ffff4eb6bda libQt5Widgets.so.5`QPushButton::paintEvent(QPaintEvent*) [inlined] QStylePainter::drawControl(opt=0x00007fffffff9bc0, ce=CE_PushButton, this=0x00007fffffff9ba0) at qstylepainter.h:85:24
    frame #24: 0x00007ffff4eb6bb8 libQt5Widgets.so.5`QPushButton::paintEvent(this=0x00005555585a7d20, (null)=<unavailable>) at qpushbutton.cpp:455:18
    frame #25: 0x00007ffff4d2de90 libQt5Widgets.so.5`QWidget::event(this=0x00005555585a7d20, event=0x00007fffffffa1c0) at qwidget.cpp:8826:19
    frame #26: 0x00007ffff4dfe214 libQt5Widgets.so.5`QAbstractButton::event(this=0x00005555585a7d20, e=0x00007fffffffa1c0) at qabstractbutton.cpp:1001:26
    frame #27: 0x00007ffff4eb83d3 libQt5Widgets.so.5`QPushButton::event(this=0x00005555585a7d20, e=0x00007fffffffa1c0) at qpushbutton.cpp:718:34
    frame #28: 0x00007ffff4ce1813 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(this=0x0000555555939d20, receiver=0x00005555585a7d20, e=0x00007fffffffa1c0) at qapplication.cpp:3640:31
    frame #29: 0x00007ffff4cea245 libQt5Widgets.so.5`QApplication::notify(this=0x00007fffffffd170, receiver=0x00005555585a7d20, e=0x00007fffffffa1c0) at qapplication.cpp:3590:31
[... cut ...]
    frame #108: 0x00007ffff3429d90 libc.so.6`__libc_start_call_main(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98) at libc_start_call_main.h:58:16
    frame #109: 0x00007ffff3429e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #110: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)
```

</details>

The crash occurs when loading a standard SVG, `dialog-ok` from `buttonBox` located in `/src/Gui/DlgPreferences.ui`.

xtemp09, 2024-02-12:
Some topics on the forum and a pull request are devoted to switch to _Qt 6_. The functions listed in the backtrace have undergone changes in transition to _Qt 6_, some of them in the range of [6.5.3, 6.7.0 rc1]. Can anyone compile _FreeCAD_ against _Qt 6_ to check if the issue persists?

xtemp09, 2024-02-24:
> Thank you! I will implement above (and then happily start using Qt6 builds).

@3x380V, sorry to mention you in this issue, but could you reproduce this crash with Qt 6?

3x380V, 2024-02-24:
@xtemp09, I've tried current git main with both Qt5 and Qt6, but cannot reproduce. Note, that only Qt5 build is using Shiboken in my case.

xtemp09, 2024-02-24:
Thank you. It's a Heisenbug. I can reproduce it by clicking the help button and Start WB options in the tree. Once in four-five tries, FreeCAD crashes.

3x380V, 2024-02-26:
I tried again. Now sure how many times, but it hurts my fingers. No crash. Also tried official FreeCAD-0.21.2-Linux-x86_64.AppImage; FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git). It crashes once in one-five tries as you noticed:
```
malloc(): mismatching next->prev_size (unsorted)
/tmp/.mount_FreeCAKqfxHJ/AppRun: line 43: 2413665 Aborted                 ${MAIN} "$@"
```

xtemp09, 2024-02-26:
So, I conclude that Qt 6 solves the problem. What version of Qt 6 have you tried? What compiler?

3x380V, 2024-02-26:
This is Debian GNU/Linux 12 (bookworm)
```
$ qmake --version
QMake version 3.1
Using Qt version 5.15.8 in /usr/lib/x86_64-linux-gnu
$ qmake6 --version
QMake version 3.1
Using Qt version 6.4.2 in /usr/lib/x86_64-linux-gnu
$ gcc --version
gcc (Debian 12.2.0-14) 12.2.0
```
Also note that Qt5 builds are using
```
FREECAD_USE_PYSIDE:BOOL=ON
FREECAD_USE_SHIBOKEN:BOOL=ON
```
while for Qt6 builds these are OFF; PySide is pip installed here.

Tried to build both versions, Qt5 version crashed after 34 clicks (Help->Start counted as single click) while Qt6 version survived 100 clicks.

xtemp09, 2024-06-25:
This issue is no longer relevant.

closed 2024-07-01 by maxwxyz