FreeCAD/FreeCAD #9952: [Problem] Fillet error

Pendelf, 2023-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33593 (Git)
Build type: Release
Branch: master
Hash: 15371a04f38bbeb5a04edc5a91020ada1e30ef88
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus 0.4.60m
  * Assembly3 0.12.0
  * Assembly4 0.50.1
  * BIM 2021.12.0
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * Design456
  * dodo 1.0.0
  * ExplodedAssembly
  * fasteners 0.4.56
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons
  * Reinforcement
  * sheetmetal 0.2.63
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

The width of the rectangle is 50 mm, and an attempt to fillet the radius of 25 mm on both sides produces an error.

### Anything else?

https://youtu.be/vxNcdgELGE8

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-07-18:
From [wiki.freecad.org](https://wiki.freecad.org/PartDesign_Fillet):

> Fillets cannot completely consume the adjacent faces.

It's a known limitation. You have to set 24.99 mm.

closed 2023-07-18 by Roy-043

Pendelf, 2023-07-18:
> From [wiki.freecad.org](https://wiki.freecad.org/PartDesign_Fillet):
> 
> > Fillets cannot completely consume the adjacent faces.
> 
> It's a known limitation. You have to set 24.99 mm.

Okay, but why is it not a problem in other programs and you can completely round

xtemp09, 2023-07-18:
Your computer ― your rules. Use programs you deem fit.

Roy-043, 2023-07-18:
The friendlier answer:
FreeCAD uses the [OCC](https://wiki.freecad.org/OpenCASCADE) CAD kernel. It is this kernel that imposes this limitation.