FreeCAD/FreeCAD #9953: [Problem] Circle centerlines don't recomputed

vladtcvs, 2023-07-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.4, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Steps to reproduce:

1. Add circle centerlines for some arc
2. Change arc center
3. Centerlines don't change their position

[image]

On this example centerlines should be in the center of arc, which has radius specified

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

vladtcvs, 2023-07-18:
> Are you using https://wiki.freecad.org/TechDraw_ExtensionCircleCenterLines?

Yes

vladtcvs, 2023-07-18:
> Please try to reproduce on development build (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) and report back. TIA

The same result

edi271, 2023-07-25:
> Are you using https://wiki.freecad.org/TechDraw_ExtensionCircleCenterLines? I don't think that function is smart enough to know that the circle has changed. It just creates a cosmetic edge based on the circle geometry at creation time. @edi271 any comments?
Correct. I don't know any method to change a cosmetic line. For example the extend/shorten line tool deletes the old line and creates a new one.

luzpaz, 2023-08-16:
@WandererFan is this a feature request then ?

luzpaz, 2023-09-07:
@WandererFan ping

WandererFan, 2023-09-07:
on the list....
...
    - hatching is only on 1 face? there was a fix for this  *** fixed
- cosmetic circle by center & radius #10352
- Annotations
...

fra589, 2023-10-30:
Same problem on my dev version discussed here: https://forum.freecad.org/viewtopic.php?t=82182

With Segmentation fault crash on Version: 0.22.0dev.34523 (Git) 

No crash on version 0.22.0dev.34789 (Git), but error messages: 
```
DVP::getProjVertexByIndex(11) - invalid index
DVP::getProjVertexByIndex(13) - invalid index
CL::scaledGeometry - failed to calculate endpoints!
CE::addCenterLinesToGeom - scaledGeometry is null
DVP::getProjVertexByIndex(12) - invalid index
CL::scaledGeometry - failed to calculate endpoints!
CE::addCenterLinesToGeom - scaledGeometry is null
<Exception> Dimension Reference has null geometry
<Exception> Dimension Reference has null geometry
```

When moving the circle object reference.

@++;
Gauthier.

luzpaz, 2023-11-17:
@WandererFan what is the proposed fix here. Perhaps another dev would step up with a fix this if they were provided orientation ?

luzpaz, 2023-11-30:
Looks related to downstream arch bug: https://bugs.archlinux.org/task/73944

WandererFan, 2023-11-30:
The center lines drawn by the extension pack are not related to the center marks that are attached to every circle/arc.   The center marks are the ones that don't appear in some linux versions.

closed 2025-01-27 by github-actions[bot]