FreeCAD/FreeCAD #9959: [Problem] Broken link for vulnerability reporting

sandr01d, 2023-07-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
-
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

The link for reporting vulnerabilities on the [security page](https://github.com/FreeCAD/FreeCAD/security) does not work, the webpage it points to does not exist.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-07-19:
Can you post a screenshot of what you see on the security page? I suspect that project managers see different things from other users, so I can't see the bad link.

sandr01d, 2023-07-19:
This is what the page looks like for me:

[image]

Clicking the link at the bottom brings me (https://github.com/FreeCAD/FreeCAD/security/advisories/new) here:

[image]

chennes, 2023-07-19:
OK, must be a permission setting I need to change. I'll track it down, thanks for the report.