FreeCAD/FreeCAD #9962: [Problem] Fatal error when opening FreeCAD from command prompt with python script

SimoneMaffei98, 2023-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

I have a python script that is used to create a 3D model in FreeCAD and then export it in .step starting from a project with a different extension.
The problem is that while running the script in python I get the following error: 'Both points are equal'.
For us this error is not a problem and can be bypassed, it just won't create the target segment.

If I run the script manually, opening FreeCAD and inserting the code in the console, FreeCad prints the error on the screen but continues to run the rest of the script (this is the right behavior).

[image]

But if I run the script automatically, through the command line when FreeCAD encounters that error it shows it on screen in the report section and stops processing the rest of the script (this is the wrong behavior).
This is the command line I use to do it automatically:
> FreeCAD.exe "" "C:\Lavoro\01-Commesse\2021\346.21.000_POLYWELD PANEL LINE FINC-RT\Software\Simonini\scripts\ElabFinCAD 04f_rel17072023 Step.py"

[image]

is it possible to solve this problem? and avoid interrupting the script when starting FreeCAD automatically from command prompt?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bdieterm, 2023-07-22:
Hi @SimoneMaffei98!
I think there is a misunderstanding.

When running a Python script, it is normal that the execution is interrupted on an unhandled error (see [here in the Python tutorial](https://docs.python.org/3/tutorial/errors.html)). When you run a FreeCAD macro, it will also be interrupted on the first unhandled error.

The FreeCAD Python Console is not for running scripts that way. When you insert your code lines in FreeCAD's Python Console via Copy&Paste, then each code block (a single line or a single line followed by indented lines) is executed separately. So you are running not a single script with 100 lines but 100 single-line scripts and after every execution of a script, you see the result or the error that made it abort the execution.

You can try this yourself:
Run this code in the Python Console and you will get your expected results -- everything will be executed even after previous errors.
```
print(1)
x1 = 1 / 0
print(2)
x2 = 2 / 0
print(3)

```

Now, let's modify the code a bit, so that it is still the same:
```
if True:
    print(1)
    x1 = 1 / 0
    print(2)
    x2 = 2 / 0
    print(3)

```
If you run this again in the FreeCAD Python Console, then only the first `1` will be printed. This is only one code block that is executed, so one small script that behaves normally and aborts on the first division by zero.

Another example: The following code is absolutely valid Python code, but it will not execute in FreeCAD Python Console (Error invalid syntax).
```
if True:
    print(1)
print(2)
```
This is because you cannot run two code blocks in the FreeCAD Python Console in one execution. If you insert an empty line before the last line, it will work again, because then it is two separate code blocks.

So this is not a bug. If you just want to automatically execute each line separately, then there are ways to do it, depending on how your scripts look like (e.g. only single line commands).

benj5378, 2023-07-22:
I 100% agree with @bdieterm that this is how Python works, that it is intended behaviour, and that is not related to FreeCAD itself. @SimoneMaffei98, if you want the script to keep running, you should implement a `try` clause that can handle the error. Feel free to post in forum.freecad.org or tag me (bensay) on FreeCAD Discord, if you need further help.

@luzpaz can you close this bug?

closed 2023-07-22 by chennes

SimoneMaffei98, 2023-07-26:
Hello @benj5378 and @bdieterm, sorry for the delay in my reply,
This is how the script looks like:
[image]
[image]

While executing these lines one could create the described error:
[image]

I cannot create a try catch for each line because I could have thousands of lines, is there a simple way to avoid the error without insert try catch for each line?

@bdieterm you said:
'So this is not a bug. If you just want to automatically execute each line separately, then there are ways to do it, depending on how your scripts look like (e.g. only single line commands).'
How could I do it? How is it possible to insert single line commands through console?

Thank you very much.

benj5378, 2023-07-26:
@SimoneMaffei98 plz attach as file. Also all the functios are collapsed.

SimoneMaffei98, 2023-07-26:
Yes, this is the script
[script.txt](https://github.com/FreeCAD/FreeCAD/files/12172629/script.txt)

benj5378, 2023-07-26:
?????? Why was there `QtCore.QCoreApplication.instance().exit()` at the end of the script??? No wonder it crashed. Why you bully me?

Here is the script with fuckit, that works: [fuckit.txt](https://github.com/FreeCAD/FreeCAD/files/12173993/fuckit.txt)

SimoneMaffei98, 2023-07-26:
Hi,
Sorry, no you are not getting pranked. The script is used to convert a 3D file to .step.
We have a software that starting from a 3D file (unknown extention) creates this python file that allows us to generate the 3D .step thanks to FreeCAD.
I know that this script works, I already used it and it works, I have only problem when I have the following error:
[image]

I am not bullying you, I use 'QtCore.QCoreApplication.instance().exit()' to close FreeCAD after the .step export.

benj5378, 2023-07-26:
@SimoneMaffei98 I believe you get this error, because the export fails due to errors in the 3D model. What is this software called, that you use to generate the Python scripts?

SimoneMaffei98, 2023-07-26:
It is another script in python but we don't have the source code.
I know that it work because we can convert files that does not have that error.

benj5378, 2023-07-26:
@SimoneMaffei98 Yes, you can convert file that does not have errors. But this script makes some error (most likely not caused by FreeCAD) and therefore the export fails. So while you can use fuckit to make as much of the model as possible, then it will still contain geometry errors.

Let's expect the export error that you encounter "Both points are equal": https://forum.freecad.org/viewtopic.php?p=480885

We can see that this is due to line 732 (in the version I sent). A quick inspection shows, that this is due to your script trying to create a line segment of lenght 0 - and that causes geomtry errors. This is not a FreeCAD error, but an error by the author of the script:
```
print(H296)  # Vector (9205.5, 675.3, 300.0)
print(H297)  # Vector (9205.5, 675.3, 300.0)
HL297=Part.LineSegment(H296,H297)
```

So whether your script usually works is unimportant. We just know, that in this specific case, the script is not working, as the your script (and not FreeCAD) are causing an error by making a line segment of length 0. This might be an issue with your converter script, it might be an issue of the 3D file you are feeding it with.

SimoneMaffei98, 2023-07-26:
Okay, fuckit is working well, thank you very much for your help.