FreeCAD/FreeCAD #9963: [Problem] Inch units are represented inconsistently `in` or `"`

Roy-043, 2023-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33593 (Git)
Build type: Release
Branch: master
Hash: 15371a04f38bbeb5a04edc5a91020ada1e30ef88
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Problem description

Inch units are represented inconsistently. For zero values `in` is used, for other values `"`.

Steps to reproduce:

In the preferences change the Unit system to "US customary".

Paste this in the Python console:
`FreeCAD.Units.Quantity(0, FreeCAD.Units.Length).UserString`
Result:
`'0,00 in'`

Paste this in the Python console:
`FreeCAD.Units.Quantity(25.4, FreeCAD.Units.Length).UserString`
Result:
`'1,00 "'`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-24:
Looks related to #9552

Roy-043, 2025-01-07:
This is still an issue.