FreeCAD/FreeCAD #9964: [Problem] FreeCAD cannot be opened because the developer cannot be verified. Error on app launch on Mac iOS

lucafaggianelli, 2023-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Problem description

After installing FreeCAD on iOS I get this error when opening it:

```
FreeCAD cannot be opened because the developer cannot be verified.
```

This is happening on recent versions of iOS, i.e. I'm running iOS 13.2.1 (22D68), and can be overcome by *force* opening an app, though is not nice to see the "not verified" error when launching an app.

For users having troubles launching FreeCAD, just keep pressed the CMD key on your keyboard and then click the FreeCAD icon, then accept the warning.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-07-25:
iOS or macOS? I was not aware FreeCAD could run on iOS

Since we already know the bundles aren't signed, is it worth keeping this issue open?

lucafaggianelli, 2023-07-25:
Ah sure I meant macOS sorry for that 

Ok, up to you could be worth as a reminder for the future 🙂

chennes, 2023-07-25:
If there isn't an issue for it already we can keep this open to remind me to keep hammering away at it. I am currently trying to work out a way for me to sign those bundles, but my last test failed.

luzpaz, 2023-08-04:
Related to https://github.com/FreeCAD/FPA/issues/6

chennes, 2023-08-04:
Actually, we can close this now, the FreeCAD 0.21 Mac bundles are signed and notarized (at long last!).

closed 2023-08-04 by chennes