FreeCAD/FreeCAD #9966: Always loses response when editing sketches with arcs

ChenKe404, 2023-07-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Version: 0.20.2
OS: win10 x64
```

### Subproject(s) affected?

Sketcher

### Problem description

When I open a sketch with arcs(one ocr more), the program always loses its response when I drag the view, and I have to forced close the program.

### Anything else?

like this one 
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-22:
At minimum we need
 - a file with the issue
 - precise steps to reproduce
 
Also please test it with 0.21 builds, the chance is high, that it is already solved and it will be nothing more backported.

Roy-043, 2023-07-29:
Too much information is missing.

Closing this as "can't reproduce".

closed 2023-07-29 by Roy-043

grubertechnic, 2023-07-29:
> Closing this as "can't reproduce".

Thanks, that makes the tracker more clean, else it will dumping here maybe years and and consumes very (re)view unnecessary lifetime. I really appreciate this.