FreeCAD/FreeCAD #9968: [Problem] RichTextAnnotation loss font settings when exporting a TechDraw model in .SVG format

DEL-TRONIK, 2023-07-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33617 (Git) AppImage
Build type: Release
Branch: master
Hash: ae215724dd6c5336dcbb263733c6830a24917963
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.4
  * Defeaturing 1.2.0
  * ThreadProfile 1.87.0
  * fasteners 0.4.56
```

### Subproject(s) affected?

Techdraw

### Problem description

In the TechDraw workshop, when using the 'RichTextAnnotation' function, the associated font settings and line break is lost when exported model in .SVG format. However, it works by exporting in .PDF format.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-07-22:
I'll need step by step instructions.  If get this result:
[image]

benj5378, 2023-07-22:
Could reproduce
1. Make new rich text format with different fonts/headings/fontsizes
2. Export as SVG

Though, this smells a bit like a Qt bug?

DEL-TRONIK, 2023-07-23:
Here are the steps that create the bug:

**1/** create a new TechDraw page

**2/** add a 'Rich Text Annotation' object with the value

    - my first line
    - my second line
    - my third line
    - my fourth line

**3/** select page, then menu File -> Export with Technical Drawing (.SVG)

Simply doing this creates the problem here, I do absolutely nothing else (no fonts or fontsize change). I've tried changing the font or size, and the problem is identical.

I don't know why it doesn't behave the same way in your home?

grubertechnic, 2023-07-23:
Can confirm here

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33631 (Git)
Build type: Release
Branch: master
Hash: 2167fb6f7ec6ac5c2f6365d4c3035d2db6879b1e
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

But it looks different, see

[image]

WandererFan, 2023-07-23:
OK, what am I doing differently?
[image]

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33631 (Git)
Build type: Unknown
Branch: master
Hash: 2167fb6f7ec6ac5c2f6365d4c3035d2db6879b1e
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/Canada (en_CA)
Installed mods: 
  * dodo 1.0.0
  * BOLTSFC 2022.11.5
  * sheetmetal 0.2.63
  * ExplodedAssembly
  * fasteners 0.4.56
  * FreeCAD_Electric
  * freecad.gears 1.0.0
  * Assembly4 0.50.2
  * A2plus 0.4.60n
[/code]

benj5378, 2023-07-23:
@WandererFan Nothing. You're just not cool enough for bugs.

Can you please supply the value of Anno Text (data field)?

WandererFan, 2023-07-23:
The AppImage seems to be rev 33617  Jul 10, 2023 which does not include f83a3fbb6a Jul 13, 2023.  f83a3fbb6a corrects the font size on Svg export and is included in master, which is at rev 33617.

This explains @DEL-TRONIK issue, but I don't know why @grubertechnic has this issue with rev 33631.

grubertechnic, 2023-07-23:
@WandererFan : i do not know either. Tomorrow i will do a full fresh compile from scratch, maybe some old things where hanging on ccache. But at least your Qt version is an other then mine.

WandererFan, 2023-07-23:
"Can you please supply the value of Anno Text (data field)?"

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN" "http://www.w3.org/TR/REC-html40/strict.dtd">
<html><head><meta name="qrichtext" content="1" /><style type="text/css">
p, li { white-space: pre-wrap; }
</style></head><body style=" font-family:'Ubuntu'; font-size:10pt; font-weight:400; font-style:normal;">
<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-family:'osifont'; font-size:12pt;">small text</span></p>
<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-family:'osifont'; font-size:18pt;">medium text</span></p>
<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-family:'osifont'; font-size:24pt;">large text</span></p></body></html>

benj5378, 2023-07-23:
@WandererFan In your screenshot, it was way longer than this?

DEL-TRONIK, 2023-07-24:
@WandererFan I've just tried it with the latest weekly-builds and the problem still seems to be there? am I missing something?

```OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git) AppImage
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.19.4
  * Defeaturing 1.2.0
  * ThreadProfile 1.87.0
  * fasteners 0.4.56```

WandererFan, 2023-07-24:
I don't think the border has ever worked correctly when exported, and I haven't figure out how to fix it yet.  

Is the text size coming out correctly for you?

[image]

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git) AppImage
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * dodo 1.0.0
  * BOLTSFC 2022.11.5
  * sheetmetal 0.2.63
  * ExplodedAssembly
  * fasteners 0.4.56
  * FreeCAD_Electric
  * freecad.gears 1.0.0
  * Assembly4 0.50.2
  * A2plus 0.4.60n
[/code]

WandererFan, 2023-07-24:
forgot to wrap the html in code tags....

```
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN" "http://www.w3.org/TR/REC-html40/strict.dtd">
<html><head><meta name="qrichtext" content="1" /><style type="text/css">
p, li { white-space: pre-wrap; }
</style></head><body style=" font-family:'osifont'; font-size:10pt; font-weight:400; font-style:normal;">
<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-size:8pt;">8 point Text</span></p>
<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-size:12pt;">12 Point Text</span></p>
<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-size:16pt;">16 Point Text</span></p>
<p style=" margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;"><span style=" font-size:20pt;">20 Point Text</span></p>
<p style="-qt-paragraph-type:empty; margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px; font-size:20pt;"><br /></p></body></html>
```

benj5378, 2023-07-24:
> ```
> <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.0//EN" "http://www.w3.org/TR/REC-html40/strict.dtd">
> <html><head><meta name="qrichtext" content="1" /><style type="text/css">
> p, li { white-space: [...]

@WandererFan Seems like we found what makes the bug occur. There is something to the generated HTML, that makes it export differently. If I copy this, I get the expected font sizes at export.

DEL-TRONIK, 2023-07-24:
yes, in fact this happens in the editor, for example if you type 'Shift + Enter' to make a line break, it will be added as a '<span>' tag, whereas if you simply use 'Enter' to make a line break, it will be added as a '<p>'. <span> will cause problems while the <p> its ok. but I'm still having position problems here and font not identical to techdraw (unless you specify the font to be used manually in the editor). I never get the same result when exporting in Techdraw to SVG.

WandererFan, 2023-07-24:
" @WandererFan There is something to the generated HTML, that makes it export differently. If I copy this, I get the expected font sizes at export. There's probably something about what html renders in the Qt documenation"

The html is correct, but after it gets rendered by QSvgGenerator the text in the output Svg is way too big.   The graphic bits are drawn the right size, but not the text.

benj5378, 2023-07-24:
> QSvgGenerator

@WandererFan That is the keyword. QSvgGenerator, not QHtmlGenerator. It's probably because the only your markup is valid in terms of svg.

grubertechnic, 2023-07-24:
fresh master build from scratch, still the same.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33651 (Git)
Build type: Release
Branch: master
Hash: e5b66905350af4c4281b2a92206720f9db57a113
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

DEL-TRONIK, 2023-07-25:
"I think this is font substitution. Is osifont installed on your system? It doesn't have to be installed for FC, but if it isn't installed, then Svg will try to find a match which might come out as ubuntu or liberation or ???"

Indeed, osifont is missing on my system (Debian), if I specify for example 'font-family:Cantarell' it works! but there are always problems of position.

grubertechnic, 2023-07-25:
> osifont is missing on my system (Debian)

It is an own package on Debian, `fonts-osifont`.

WandererFan, 2023-07-30:
This should be improved by PR #10019.

I'm not entirely happy with the algo for predicting how the text should be positioned, but it seems to work in my test cases.

DEL-TRONIK, 2023-08-07:
Sorry for the late reply, I've just tried it with the version 33719 (lastest weekly-builds) and everything now works perfectly! the border now seems to work too! it's great, thank you @WandererFan !

ps: note that even though I've installed the osi-font package for the default font, I still have to specify a font manually (via Rich text editor -> Edit document source) for everything to be positioned correctly. for example, if I use the default value (osi-font) there are always some positioning discrepancies, however if I specify cantarell (which is the font used by my debian) everything is positioned correctly.

WandererFan, 2023-08-07:
Glad it works for you.

I'm not enough of an Svg wizard to understand how/when it uses the named font and when it substitutes another.

benj5378, 2023-08-07:
When importing SVGs, but could be related https://bugreports.qt.io/browse/QTBUG-65562

closed 2023-08-20 by WandererFan