FreeCAD/FreeCAD #9973: [Problem] Crash when using subtractive helix in reversed

fwillo, 2023-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * freecad.gears 1.0.0
  * Manipulator 1.4.9
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

I have a small minimal example attached. Appling subtractive helix on Sketch001 and ticking "Reversed" crashes FreeCAD.

### Anything else?

[Crash.zip](https://github.com/FreeCAD/FreeCAD/files/12138581/Crash.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-23:
Works flawless on 0.21master. Also see here #8821, already solved.

Please retest it on the current 0.21dev version and consider to close it, when it works.

grubertechnic, 2023-07-23:
This is simply because the result is invalid. the profile co-touches the turn axis, which makes which causes floating point issues. Also the shape have an infinity gradient in the center.

[image]

That is fails is exactable. If you make an small gap between the turn axis and the profile, it works.

Note: generally atm helix have some issues (see tracker, heights incorrect >10^(-7)mm, some B-Spline imperfections), but in this file none occurs.

fwillo, 2023-07-23:
Ok, sorry for false alarm. Closing.

closed 2023-07-23 by fwillo

grubertechnic, 2023-07-23:
> sorry for false alarm

No need to be sorry.

Note: the geometric kernel of FreeCAD is limited, but works with every update better. This case is pretty on edge, where even commercial kernels can work or not. Atm i can not test it, but i am sure, that even other kernels will have here issues, maybe not on the first operation, but then later on the next few operations.