FreeCAD/FreeCAD #9980: [Problem] Linked object overridden material don't have shininess

Anthony-Gaudino, 2023-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.2.6
  * Silk 0.1.3
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.57
  * lattice2 1.0.0
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * Curves 0.6.12
  * Assembly4 0.50.2
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

When I create any object in FreeCad, if I import that object to another file as a linked object, then set it to `Override Material: true` and changes the material to for example `Metalized` in the `Appearance...` task, the object doesn't have any shininess.

Here's the linked object with the `Metalized` material applied:
[image]
It appears as a solid black object.

Here's the original object with the same material applied:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-23:
It does not work on the master either.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git)
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

But in general i am not happy with the property `Override Material`, since this is no material, only appearance. This should be changed because it is misleading.

Anthony-Gaudino, 2023-07-24:
Maybe the issue is around here: https://github.com/FreeCAD/FreeCAD/blob/a3851d4701e66c73265fafb3d16ea0dcbbd09c19/src/Gui/ViewProviderLink.cpp

I can see references for transparence but no shininess.

luzpaz, 2023-07-24:
Maybe something that overlaps with @davesrocketshop's Material revamp ?

Anthony-Gaudino, 2023-07-24:
Looking at the code and testing on FreeCad, it seems that only `diffuseColor` and transparency are overriden in the link. All other appearance attributes are ignored.

grubertechnic, 2023-07-24:
@luzpaz : not really. It is an issue in the name of the properties of the link object.

davesrocketshop, 2023-07-25:
Unrelated to this, I've noticed in other places that appearance only uses diffuse colour and transparency, including in the material editor preview. This will be reviewed as part of the material overhaul, but that is a long term project. This should probably be fixed earlier.

davesrocketshop, 2023-07-25:
Looking at the code, what's being linked is the App::Color object, which includes a color and transparency value, but not the other appearance properties. I have no idea how complicated a fix might be. I'll dig deeper tomorrow.

Anthony-Gaudino, 2024-07-12:
Updated Full version info as it still happens on 0.22.0dev.
There's still problems assigning materials to linked objects without affecting the parent.

luzpaz, 2025-01-16:
@davesrocketshop can you review the relevancy of this ticket when you have a spare moment? TIA

davesrocketshop, 2025-01-19:
Well, it's a mixed bag. When I use the "Inspect Appearance..." and "Inspect Material..." tools, they're all set correctly. But they're not displaying correctly.

Here's what I'm seeing:

[image]

Both sphere's have the same setting. The one on the right is the linked object. The same issue occurs across files

davesrocketshop, 2025-01-19:
Never mind. It's working correctly. "Override Material" should probably be changed to "Override Appearance", and I just saw where those settings live

[image]

I'd close this as no longer relavent

luzpaz, 2025-01-19:
> "Override Material" should probably be changed to "Override Appearance", and I just saw where those settings live

@davesrocketshop are you making a PR ?  If so, then you should auto-close this ticket through said PR.

maxwxyz, 2025-01-19:
Closing in favor of https://github.com/FreeCAD/FreeCAD/issues/19135

closed 2025-01-19 by maxwxyz