FreeCAD/FreeCAD #9981: [Problem] Applying constraints on external geometry causes crash

Anthony-Gaudino, 2023-07-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.2.6
  * Silk 0.1.3
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.57
  * lattice2 1.0.0
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * Curves 0.6.12
  * Assembly4 0.50.2
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

If I import external geometry in a sketch and then apply constraints on the imported geometry it causes an error in console the first time and if any other time during FreeCad runtime I apply constraints like this it causes FreeCad to crash.

This can happen if for example I import a line and apply a horizontal or vertical constraint to this line or to the points of this line.

See example:
https://github.com/FreeCAD/FreeCAD/assets/2439728/2d3b919c-9c76-4b2e-a2ee-07519171e6ba
After applying the horizontal constraint to the points twice FreeCad crashes.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-23:
Works here flawless.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git)
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

Please retest it on the newest 0.21master build. 0.20 is pretty old yet and there will be no backporting. 0.21master is already feature freezing/frozen.

luzpaz, 2023-07-24:
Closing due to fix not being backported and immanent upcoming new version release

kafolske, 2023-12-22:
Can this be re-opened?  I am see this on the most recent release on Mac:

[code]
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]

luzpaz, 2023-12-22:
FreeCAD v0.21.2 release is delayed due to macOS signing issues. @kafolske can you reproduce on latest development version (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) ?

kafolske, 2023-12-23:
Yes.  I was able to reproduce the same crash on this build.

luzpaz, 2023-12-23:
@kafolske can you paste the full [about](https://wiki.freecad.org/About) info for the revision you reproduced on?

kafolske, 2023-12-24:
@luzpaz Sorry about that.  Here it is:

```
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

bozzzzo, 2024-03-08:
Maybe a shorter reproducer:

Empty file
Sketcher workbench
create sketch
select dimension tool
click on origin point
crash

```
OS: macOS 13.6.2
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: C/Default (C) ]
Installed mods: 
  * ThreadProfile 1.89.0
  * ProDarkThemePreferencePack 1.0.0
  * fcgear 1.0.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.4.72
  * lattice2 1.0.0
  * ExtremeProDark 2.9.0
  * Curves 0.6.15

```

luzpaz, 2024-03-08:
@JohnOCFII can you confirm on macOS ?

JohnOCFII, 2024-03-09:
I am not able to reproduce a crash or error following either @bozzzzo or @Anthony-Gaudino's examples.  I was able to to the following:

1) Create sketch - Create a rectangle in this sketch
2) Create 2nd sketch - Choose external geometry to bring in the top line of the first sketch
3) Using Horizontal constraint, enter distance for the top line in the new (2nd) sketch
4) Use Command-Z to remove that line, and then re-enter it. 
[image]

All worked fine.

```OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
  ```

JohnOCFII, 2024-03-09:
As noted above (and in one other similar issue) I'm running the now-current macOS version and no longer see the issue.  When I DID see the issue previously, I was running the previous version of macOS (not that FreeCAD should crash on the most recent previous major version -- but it is a difference I noted).

Anthony-Gaudino, 2024-03-09:
I can confirm that I don't have this error anymore.
No error appear on console and no crash happens.

My current full version info:
```
OS: macOS 14.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.22
  * ThreadProfile 1.89.0
  * Launcher
  * Nodes 0.1.36
  * InventorLoader 1.4.0
  * woodworking 0.21.2.33771
  * Render 2024.1.23
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * nurbs
  * A2plus 0.4.64
  * Manipulator 1.5.7
  * fasteners 0.5.12
  * lattice2 1.0.0
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.2
  * toSketch 1.0.1
  * Curves 0.6.23
  * Assembly4 0.50.7
  * Assembly3 0.12.2
```

Between the first time I reported the issue, I have updated the OS, FreeCAD and Python.

bozzzzo, 2024-03-10:
I upgraded the OS to 14.4 and now I cannot reproduce anymore

FEA-eng, 2024-06-29:
@maxwxyz Based on the replies above, it seems that this issue can be closed.

closed 2024-06-29 by maxwxyz