FreeCAD/FreeCAD #9990: [Problem] TechDraw page returns incorrect width and height in Python

benj5378, 2023-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Ubuntu 22.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33650 (Git)
Build type: Debug
Branch: iss11
Hash: 0fee9c98f036d18762b6eceb4a6aecbc8f9f948d
Python 3.10.7, Qt 5.15.6, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Techdraw

### Problem description

If one tries to retrieve the page width and height before the TechDraw page has been opened in the gui, then one gets the value `0`.

Example code from @Roy-043 

```
import FreeCAD as App

doc = App.ActiveDocument
box = doc.addObject("Part::Box", "Box")

page = doc.addObject("TechDraw::DrawPage", "Page")
template = doc.addObject("TechDraw::DrawSVGTemplate", "Template")
template.Template = App.getResourceDir() + "Mod/TechDraw/Templates/A4_LandscapeTD.svg"
page.Template = template

view = doc.addObject("TechDraw::DrawViewPart", "View")
page.addView(view)
view.Source = [box]
view.Direction = (0.0, 0.0, 1.0)

doc.recompute()

page.PageWidth  // Returns 0
page.PageHeight  // Returns 0
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2023-07-24:
https://forum.freecad.org/viewtopic.php?p=695067#p695067

luzpaz, 2023-08-17:
What's the status on this ?

WandererFan, 2023-08-18:
It is build on top of a PR that doesn't work.

WandererFan, 2023-08-20:
Should be fixed by commit 622df46971.

closed 2023-08-20 by WandererFan

benj5378, 2023-08-20:
@WandererFan Is #9991 still relevant?

benj5378, 2023-08-23:
@WandererFan Is https://github.com/FreeCAD/FreeCAD/pull/9991 still relevant?

WandererFan, 2023-08-24:
The part about using only 1 property for the template would be good but hard to implement while preserving backwards compatibility.   Making PageResult hidden in the tree might solve most of the confusion around having 2 properties with only a 1 line change.