FreeCAD/FreeCAD #9992: [Problem] Crash when clicking help button

benj5378, 2023-07-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.10 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33651 (Git)
Build type: Debug
Branch: iss12
Hash: e5b66905350af4c4281b2a92206720f9db57a113
Python 3.10.7, Qt 5.15.6, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Problem description

1. Install the help addon
2. Go to Help in menu bar and then press Help in the dropdown
3. Crash 😢

### Anything else?

Traceback

```
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff2d96a5d in QWidget::windowTitle() const () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
(gdb) bt
#0  0x00007ffff2d96a5d in QWidget::windowTitle() const () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007ffff66a0843 in Gui::Application::viewActivated(Gui::MDIView*) (this=0x7fffffffd2b0, pcView=0x0)
    at /home/bensay/Development/FreeCAD/src/Gui/Application.cpp:1256
#2  0x00007ffff6ea0a70 in Gui::MainWindow::onWindowActivated(QMdiSubWindow*) (this=0x7fffffffd270, w=0x55555a3827b0)
    at /home/bensay/Development/FreeCAD/src/Gui/MainWindow.cpp:1174
#3  0x00007ffff6eb2a82 in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QMdiSubWindow*>, void, void (Gui::MainWindow::*)(QMdiSubWindow*)>::call(void (Gui::MainWindow::*)(QMdiSubWindow*), Gui::MainWindow*, void**)
     (f=(void (Gui::MainWindow::*)(Gui::MainWindow * const, QMdiSubWindow *)) 0x7ffff6ea09c4 <Gui::MainWindow::onWindowActivated(QMdiSubWindow*)>, o=0x7fffffffd270, arg=0x7fffffffa710) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:152
#4  0x00007ffff6eb1927 in QtPrivate::FunctionPointer<void (Gui::MainWindow::*)(QMdiSubWindow*)>::call<QtPrivate::List<QMdiSubWindow*>, void>(void (Gui::MainWindow::*)(QMdiSubWindow*), Gui::MainWindow*, void**)
     (f=(void (Gui::MainWindow::*)(Gui::MainWindow * const, QMdiSubWindow *)) 0x7ffff6ea09c4 <Gui::MainWindow::onWindowActivated(QMdiSubWindow*)>, o=0x7fffffffd270, arg=0x7fffffffa710) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:185
#5  0x00007ffff6eb04ff in QtPrivate::QSlotObject<void (Gui::MainWindow::*)(QMdiSubWindow*), QtPrivate::List<QMdiSubWindow*>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x555556288db0, r=0x7fffffffd270, a=0x7fffffffa710, ret=0x0)
    at /usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:418
#6  0x00007ffff20f36ff in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff2ed6243 in QMdiArea::subWindowActivated(QMdiSubWindow*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff2edd29f in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff20f372c in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#10 0x00007ffff2ee1949 in QMdiSubWindow::windowStateChanged(QFlags<Qt::WindowState>, QFlags<Qt::WindowState>) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff2eeb5e4 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff2eec102 in QMdiSubWindow::eventFilter(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff20baba3 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff2d6bf22 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff68148da in Gui::GUIApplication::notify(QObject*, QEvent*) (this=0x7fffffffd230, receiver=0x55555a309210, event=0x7fffffffacb0)
    at /home/bensay/Development/FreeCAD/src/Gui/GuiApplication.cpp:85
#16 0x00007ffff20bae38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff2dab6f7 in QWidgetPrivate::show_helper() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff2dae693 in QWidgetPrivate::setVisible(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007fffd80d18d9 in  () at /usr/lib/python3/dist-packages/PySide2/QtWebEngineWidgets.cpython-310-x86_64-linux-gnu.so
#20 0x00007ffff2dab639 in QWidgetPrivate::showChildren(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff2dab6d3 in QWidgetPrivate::show_helper() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff2dae693 in QWidgetPrivate::setVisible(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007fffcbc908d0 in  () at /usr/lib/python3/dist-packages/PySide2/QtWidgets.cpython-310-x86_64-linux-gnu.so
#24 0x00007ffff352be92 in  () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#25 0x00007ffff347b407 in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#26 0x00007ffff35cb046 in  () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#27 0x00007ffff3477f1e in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#28 0x00007ffff35cb046 in  () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#29 0x00007ffff3477f1e in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#30 0x00007ffff35cb046 in  () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#31 0x00007ffff347b407 in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#32 0x00007ffff35cb046 in  () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#33 0x00007ffff35c5a0d in PyEval_EvalCode () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#34 0x00007ffff3615e69 in  () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#35 0x00007ffff3616fa9 in PyRun_StringFlags () at /lib/x86_64-linux-gnu/libpython3.10.so.1.0
#36 0x00007ffff3d2c83c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) (this=0x555555636a30, sCmd=0x555559192668 "Help.show(\"Online_Help_Startpage\")")
    at /home/bensay/Development/FreeCAD/src/Base/Interpreter.cpp:240
#37 0x00007ffff6879373 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*)
    (file=0x7ffff7976628 "/home/bensay/Development/FreeCAD/src/Gui/MainWindow.cpp", line=854, eType=Gui::Command::Gui, sCmd=0x555559192668 "Help.show(\"Online_Help_Startpage\")")
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:726
#38 0x00007ffff6878fc7 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...)
    (file=0x7ffff7976628 "/home/bensay/Development/FreeCAD/src/Gui/MainWindow.cpp", line=854, eType=Gui::Command::Gui, sCmd=0x7ffff797628a "Help.show(\"%s\")")
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:678
#39 0x00007ffff6e9ed7b in Gui::MainWindow::showDocumentation(QString const&) (this=0x7fffffffd270, help=...) at /home/bensay/Development/FreeCAD/src/Gui/MainWindow.cpp:854
#40 0x00007ffff68ba43b in StdCmdOnlineHelp::activated(int) (this=0x555555baf0a0, iMsg=0) at /home/bensay/Development/FreeCAD/src/Gui/CommandStd.cpp:486
#41 0x00007ffff6877d17 in Gui::Command::_invoke(int, bool) (this=0x555555baf0a0, id=0, disablelog=true) at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:445
#42 0x00007ffff6877a4f in Gui::Command::invoke(int, Gui::Command::TriggerSource) (this=0x555555baf0a0, i=0, trigger=Gui::Command::TriggerAction)
    at /home/bensay/Development/FreeCAD/src/Gui/Command.cpp:410
#43 0x00007ffff685ede2 in Gui::Action::onActivated() (this=0x555556a4f920) at /home/bensay/Development/FreeCAD/src/Gui/Action.cpp:112
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-07-25:
does it also crash with a release build?

adrianinsaval, 2023-07-25:
I meant with build type = Release but still using 0.21

benj5378, 2023-07-26:
Yes, still crash https://discord.com/channels/870877411049357352/870877411623985154/1133729681946988545

benj5378, 2023-07-26:
@chennes If we want a hot fix for 21.0, then set default setting for help page to open in browser tab. Idk why this isn't just the default already, who wants to navigate web pages inside FreeCAD?

adrianinsaval, 2023-07-26:
it's probably useful for sandboxed environments like flatpak or snap

chennes, 2023-07-26:
@yorikvanhavre Help is really your area of expertise: any idea what's going on here?

yorikvanhavre, 2023-07-27:
Ok got it... it happens when "in a new FreeCAD tab" is selected under edit->preferences->help. Investigating...

yorikvanhavre, 2023-07-27:
```python
from PySide import QtGui
m = FreeCADGui.getMainWindow()
mdi = m.findChild(QtGui.QMdiArea)
s = QtGui.QMdiSubWindow()
s.setWidget(QtGui.QCalendarWidget()) # not sure this is even needed...
mdi.addSubWindow(s)
mdi.setActiveSubWindow(s) # <-- this does not crash but freezes the tabs
s.show() # <-- this causes a crash
```

It shouldn't cause a crash... It seems like Qt's fault, but I'll investigate more.

yorikvanhavre, 2023-07-27:
A workaround would be to use the Web module's openBrowserHTML... The problem is that we have no way to handle clicked links there, and we need this to render markdown. A workaround to this workaround would be to display the corresponding github page instead of the markdown version

yorikvanhavre, 2023-07-31:
The above commit solves the crashes, but any clicked link on the doc page is not rendered in HTML. Setting up github pages to have a workaround against that...

luzpaz, 2023-08-11:
What's the status of this ticket ?

yorikvanhavre, 2023-09-11:
Still not done but I'm working on it.

luzpaz, 2023-09-23:
@yorikvanhavre can we link the gsoc doc student to this ticket, for testing ?

luzpaz, 2023-10-24:
@yorikvanhavre is this still an issue after the gsoc merged code ?

yorikvanhavre, 2023-10-27:
No it's solved

closed 2023-10-27 by yorikvanhavre