FreeCAD/FreeCAD #9996: [Problem] Language change doesn't work properly

TuomasLattila, 2023-07-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Finnish/Finland (fi_FI)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Test case:

1. Open freeCAD
2. Go to Edit > Preferences ... > Change language
3. Change language to different one
4. Press OK

Now the language should be changed, but actually some of the text (Start page) are written still in the previous language. 

To get the Start page texts also in the right language you must restart freeCAD.

### Anything else?

[image]

Here is example picture where the language has been changed from Finnish to English.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-07-25:
IF the translation is the issue, there is an extra tracker for it, see https://github.com/FreeCAD/FreeCAD-translations/issues. Also maybe @kaktusus know more about that.

kaktusus, 2023-07-25:
I tested on new builds

```
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git) AppImage
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
...
```

After changing the language in the preferences, the start page is not refreshed and the tab labels are not reloaded.
Therefore changing the language is not 100% effective.
A restart of FreeCAD is necessary in this case.

Please move this ticket to FreeCAD    /    FreeCAD-translations.

TuomasLattila, 2023-07-25:
Yes, that is the bug that I tried to show in this ticket. It does the translation when restarted.

What is the best way to move this ticket? I have never done it before.

kaktusus, 2023-07-25:
This must be done by a person with higher authority.

I recommend you to use the developer version. There are a lot of improvements there and you have access to new features.

You can download it here:

https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#Asets
You download, unzip, and run it
and enjoy the freshness. :grinning: 
Nothing needs to be installed.

A developer once asked me to report such cases when a reboot is required to update the GUI language. But I forget who it was. :cry:

TuomasLattila, 2023-07-25:
Ah ok, I don't have the authority. Thanks for the recommendation :+1:

benj5378, 2023-07-26:
Is this really a bug or just a technical limitation of how FreeCAD works? I believe it would take an extrodinary workload and add uneeded complexity to make sure all gui is reloaded without restarting FreeCAD... when one can just restart FreeCAD

kaktusus, 2023-07-26:
I don't consider it a bug. :wink: 
But as I mentioned earlier some developer asked me to catch such "phenomena". Apparently he thinks otherwise.
As for me, it is worth moving it to the translation corner.

yorikvanhavre, 2023-07-26:
There is a way to do this relatively simply, on [Start workbench Activated/Deactivated](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Start/InitGui.py#L43C37-L43C37), an observer must be added/removed to the language property, that triggers a reload. Something like this (not tested):

```python
class observer:
    def onChange(self, paramGrp, param):
        if(param == "Language"):
            mw = FreeCADGui.getMainWindow()
            if mw:
                # close all opened start page instances
                title = QtGui.QApplication.translate("Workbench","Start page")
                mdi = mw.findChild(QtGui.QMdiArea)
                if mdi:
                    for mdichild in mdi.children():
                        for subw in mdichild.findChildren(QtGui.QMdiSubWindow):
                            if subw.windowTitle() == title:
                                subw.close()
            FreeCADGui.runCommand("Start_StartPage")

p = FreeCAD.ParamGet("User parameter:BaseApp/Preferences/General")
p.Attach(observer())
```
This observer must then be removed on Start WB deactivation, so it does not weight on normal FreeCAD operation

yorikvanhavre, 2023-07-26:
It is not exactly a translation issue, though... Isn't it better to leave it here?

kaktusus, 2023-07-26:
Such topics were discussed in that corner .... well unless I misremembered.

do as you think  :wink:

kaktusus, 2023-07-26:
Yorik please add a wiki editor account for a girl from the Polish team.
Sorry to bring this up here but it's important to me.

yorikvanhavre, 2023-07-26:
which girl?

kaktusus, 2023-07-26:
Kat_ka

yorikvanhavre, 2023-07-27:
Done!

luzpaz, 2023-07-28:
#6704

kaktusus, 2023-07-28:
Thanks luzpaz for finding this topic.

kaktusus, 2023-07-28:
However, I misremembered the conversation about the necessary restart was going on in the main branch.

yorikvanhavre, 2023-07-28:
I think we can add the above change nevertheless, but it's better to do it after 0.21 as it could introduce new bugs.

luzpaz, 2023-08-04:
@yorikvanhavre should this be assigned to you ?

yorikvanhavre, 2023-08-07:
Yes!

superbonaci, 2023-11-10:
Same happens to cube that describes the views: TOP, LEFT, FRONT, etc. A restart is required to match the language of the words in the faces the language in the menu:

[image]

The language has been changed from English to French but some parts are not translated until restart. Maybe just adding to preferences that needs restart would be enough:
`Preferences` - `General` - `Language (needs program restart):`

Is important to mention that the restart if for the program and not for the OS.

kaktusus, 2023-11-10:
Since this topic returns like a boomerang I will remind you of the possibility of its solution:

https://github.com/FreeCAD/FreeCAD/issues/6704#issuecomment-1716986923

:wink:

superbonaci, 2023-11-10:
> Since this topic returns like a boomerang I will remind you of the possibility of its solution:
> 
> [#6704 (comment)](https://github.com/FreeCAD/FreeCAD/issues/6704#issuecomment-1716986923)
> 
> 😉

The solution is pretty simple: mention that a program restart is required inside Settings.

superbonaci, 2023-11-16:
I think this is the line that has to be changed:
https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/PreferencePages/DlgSettingsGeneral.ui#L41

Also the drop list has to be moved to give way for translations:

[image]

I could submit a pull request if somebody tells me which code to change.

yorikvanhavre, 2023-11-20:
I just had a look at this, the start page is apparently well reloaded when the language changes. The problem is that the strings in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Start/StartPage/TranslationTexts.py get translated once, when the module is imported, at FreeCAD startup, and not anymore after that.

The solution here is probably to:

1. Place all translated strings into some retranslate() function
2. One needs to decide how StartPage.py will access these strings. Either they are global variables and get updated by the retranslate function, or the StartPage module needs to fetch them differently
3. Run that function everytime the start page is generated

kaktusus, 2023-11-20:
I know that the developers decided to make such functionality possible. But as you can see there are more problems with it than benefits. In addition there is no one to take care of it.

Is this really necessary? Maybe the information about the need to reload the program will be enough?

yorikvanhavre, 2023-11-20:
The problem is that adding a message is just a cosmetic workaround, and it will need an extra commit to remove it afterwards. It's better to fix things correctly right from the start.

I will do this when I have a moment, but my time is limited right now

luzpaz, 2023-11-27:
Can y'all test yorik's fix ?

superbonaci, 2023-11-27:
> Allow dynamic translation of start page -
can't compile due to https://github.com/FreeCAD/FreeCAD/issues/11438

superbonaci, 2023-11-27:
> Can y'all test yorik's fix ?

Can't compile due to https://github.com/FreeCAD/FreeCAD/issues/11438

closed 2023-11-27 by chennes

kaktusus, 2023-11-28:
It agrees the cube wall captions after switching the language were not refreshed correctly.

kaktusus, 2023-11-28:
https://crowdin.com/translate/freecad/549/en-pl#6613111
https://crowdin.com/translate/freecad/549/en-pl#6613109
https://crowdin.com/translate/freecad/549/en-pl#6613107
https://crowdin.com/translate/freecad/549/en-pl#6613105
https://crowdin.com/translate/freecad/549/en-pl#6613103
https://crowdin.com/translate/freecad/549/en-pl#6613101

superbonaci, 2023-11-28:
> https://crowdin.com/translate/freecad/549/en-pl#6613111 https://crowdin.com/translate/freecad/549/en-pl#6613109 https://crowdin.com/translate/freecad/549/en-pl#6613107 https://crowdin.com/translate/freecad/549/en-pl#6613105 https://crowdin.com/translate/freecad/549/en-pl#6613103 https://crowdin.com/translate/freecad/549/en-pl#6613101

It requires an account to see the content.

superbonaci, 2023-11-28:
Thanks but not needed. Is there where translations are made?

kaktusus, 2023-11-28:
Yes, we do translations on the Crowdin platform. That's where these links come from.
here's how it looks:

[image]

PR #11513 (2023-11-27): Start: Allow dynamic translation of start page
- reloading the start page now retranslates all the texts
- extend the observer to watch language changes

Fixes #9996
commit d565796f (2023-11-27): Start: Allow dynamic translation of start page (#11513)

* Allow dynamic translation of start page - fixes #9996
* [pre-commit.ci] auto fixes from pre-commit.com hooks