FreeCAD/FreeCAD #10001: [Problem] Recent Files is weirdly positioned in the menu.

PaddleStroke, 2023-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
0.21
```

### Subproject(s) affected?

Core

### Problem description

[image]

Recent files is weirdly placed by itself at the end of the menu.
I suggest renaming it 'Open recent' and place it right under Open.
This will make it more intuitive and make the menu easier on the eye : remove one separator and one section. And naming it 'Open ----' makes the eye instantaneously understand what this is, which makes the menu look less cluttered.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-07-28:
I've just opened Origin, it has _Recent_ at the bottom of the menu, a near _Exit_. So, as LibreCAD, Notepad++, Diamond.

Though, I have software that has _Recent_ a near _Open_, Inkscape and Paint.NET.

So far I concluded that programs put it at the bottom of the file menu, except for graphics editors.

PaddleStroke, 2023-07-29:
I have looked at a lot of softwares, and found some that put it at the bottom indeed. However not only graphic editors put it near open. Few other examples : Eagle, LibreOffice, VLC, KDenLive...

Putting recent files at the end makes sense in the case where they are directly in the menu :
[image]
Because then the menu stays 'static' but the end of the menu that changes with the files.

But in the case where it is only 'Recent Files' submenu put at the end, I cannot see any advantage. Whereas putting it as Open Recent next to Open increase clarity.

closed 2023-07-30 by PaddleStroke