FreeCAD/FreeCAD #10006: [Problem] PartDesign have issues, when no Sketch is in the document, which is deprecated workflow

grubertechnic, 2023-07-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33657 (Git)
Build type: Release
Branch: master
Hash: 2a1d4d4994bf3707f7a07aef07e17f5270b4854b
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * CurvedShapes 1.0.4
  * Curves 0.6.12
```

### Subproject(s) affected?

PartDesign

### Problem description

PartDesign have issues, when no Sketch is in the document, which is deprecated workflow. Even with no sketch in the document, PartDesign works now.

Steps to reproduce:
 - load the attached file
 - try the use any PartDesign task
 - now the message `No sketch is available in the document` occur
 - create a sketch, does not matter if it is in the Body or not
 - now it works
 
[2023-07-2_22-44.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/ddf5ec65-11a8-4bb2-bfdb-55853cd040b2)

### Anything else?

Minor issue

Note: there is a selecting issue on the task dialog, i can not use faced direct. Anyway it is a separate issue and affects not this issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-25:
@wwmayer care to take a stab at this ?

wwmayer, 2023-08-25:
> PartDesign have issues, when no Sketch is in the document, which is deprecated workflow.

I don't think it's deprecated because there are valid use cases where a sketch is not needed. For an existing object it's e.g. possible to directly pad or pocket a face without having to create a sketch.

> try the use any PartDesign task

I tried a pad and this worked for me without raising an error or warning. So, which commands have you tested?

grubertechnic, 2023-08-26:
> I don't think it's deprecated because there are valid use cases where a sketch is not needed.

Yes, that is the point, maybe i have written that misleading.

> I tried a pad and this worked for me without raising an error or warning.

Without selecting anything before? I still get the message (today master fresh compiled yet d623e37e5ef269848c979915ed480ea9bbf7ce83).

> So, which commands have you tested?

Tested all and it is affecting all additive and subtractive PartDesign features except primitives, probably because they have no sketch dependencies per default. To be explicit:

additive: Pad, Revolution, Additive Loft, Additive Pipe, Additive Helix
subtractive: Pocket, Hole, Groove, Subtractive Loft, Subtractive Pipe, Subtractive Helix

Note, if you open the example file, then you get an other warning for subtractive elements, because there is no solid in the Body. When you add an additive primitive and then select for example a pocket, you still get the same warning, because there is no sketch in the file.

[2023-08-26 17-56.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/16e565c1-7960-4496-a95f-1c75fa60a5b5)

wwmayer, 2023-08-26:
> Without selecting anything before? I still get the message 

Of course, I have selected the shape binder as you nowhere wrote that nothing should be selected. But if nothing is selected then it's expected that the pad will fail because it doesn't contain a sketch and the pad cannot anticipate which face to directly pad.

So, I don't see anything wrong but the error message could be clearer.

grubertechnic, 2023-08-26:
> But if nothing is selected then it's expected that the pad will fail because it doesn't contain a sketch and the pad cannot anticipate which face to directly pad.

But that is the issue. When i add an random sketch (open example file, SketchWB, add Sketch, close it empty) somewhere in the file, independent from the Body, then it works. Also you do not need a sketch per se anymore, because like in the example, the wire (the base of the operation) is no sketch, it is a shape binder (or any other face).

Maybe i should express the issue different, without a preselection a PartDesign additive/subtractive task does not work without sketch.

I do not say, that i work without sketches, it is just cumbersome to add a sketch in a file, when it is not needed.

grubertechnic, 2023-08-26:
It is 2 issues in one.