FreeCAD/FreeCAD #10008: [Problem]  0.20.2 fails to start on Windows 10

gzcwnk, 2023-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
It wont start/open
```

### Subproject(s) affected?

None

### Problem description

I have 0.19.3 working OK so just tried to upgrade and 0.20.2 will not run while 19.3 still runs fine.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gzcwnk, 2023-07-28:
I installed it as a user and the user cannot run it.  I just tried as administrator and it runs OK.

version 0.20.2 build 29603 (Git) 

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/New Zealand (en_NZ)
[/code]

xtemp09, 2023-07-28:
@gzcwnk, could you provide the log file of the failed run? To do this you need to run FreeCAD with `-l` or `--write-log` option. [^1]

I thought installation as a user to be impossible, since FreeCAD installs some MS redistributable and sets some environment variables like `PYTHONPATH`.

[^1]: https://wiki.freecad.org/Debugging

benj5378, 2023-07-28:
This seems like an installation issue and not a bug. Can we please close this? If you need help installing/starting FreeCAD, feel free to make a post on forum.freecad.ord. You can also print the log by using `& .\FreeCAD.exe | Out-Host` in PowerShell

gzcwnk, 2023-07-28:
A failed install = a bug.

gzcwnk, 2023-07-28:
> @gzcwnk, could you provide the log file of the failed run? To do this you need to run FreeCAD with `-l` or `--write-log` option. [1](#user-content-fn-1-9291caafc23cdd1109dc838c51a7b794)
> 
> I thought installation as a user to be impossible, since FreeCAD installs some MS redistributable and sets some environment variables like `PYTHONPATH`.
> 
> ## Footnotes
> 1. https://wiki.freecad.org/Debugging [↩](#user-content-fnref-1-9291caafc23cdd1109dc838c51a7b794)

I retried as admin still failed.   

I have no idea how to use powershell

gzcwnk, 2023-07-28:
PS C:\Users\steven> & .\FreeCAD.exe | Out-Host
& : The term '.\FreeCAD.exe' is not recognized as the name of a cmdlet, function, script file, or operable program.
Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:3
+ & .\FreeCAD.exe | Out-Host
+   ~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (.\FreeCAD.exe:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

PS C:\Users\steven>

gzcwnk, 2023-07-28:
Looks like an ongoing issue for others also.

https://forum.freecad.org/viewtopic.php?t=78995

gzcwnk, 2023-07-28:
fixed by,

"Your user configuration and some other files are stored in C:\Users\*YOURUSERID*\AppData\Roaming\FreeCAD.
You can delete the entire FreeCAD folder found there."

Screams bug to me.

benj5378, 2023-07-28:
@gzcwnk A failed install by itself is not a bug if it's an error caused by the user; here it sounded like you installed FreeCAD in the wrong locaton. The forum post is a very few others having an issue; there are hundreds of people using FreeCAD every day - and they have done so for months without this issue, so it doesn't seem like a general issue.

When we talk about bugs in FreeCAD, we talk about stuff that has to be fixed within FreeCAD. Sometimes startup issues has to do with graphics drivers are bugs in third party software/libraries and are therefore not considered as bugs within here (FreeCAD). Therefore, all startup issues isn't automatically bugs.

Though when deleting this folder, causes behaviour change of the software, it might actually be a bug. When I wrote `& .\FreeCAD.exe | Out-Host`, it was meant to be run inside the installation folder, like this `PS C:\Program Files\FreeCad> & .\FreeCAD.exe | Out-Host`. It could have been interesting knowing the output of this, as it could have given clues to whether it's a bug actually inside of FreeCAD. Please supply it, if it would have to happen again.

gzcwnk, 2023-07-30:
Firstly, you are guessing and wrong, I did a default install incl location. (I learned that one long ago). 
Secondly, you provided inadequate instructions.

benj5378, 2023-07-30:
@gzcwnk I think you are misunderstanding something here. That this command should be run inside the installation folder is so basic command-line know how, that it is trivial to explain; it is not inadequate instructions in the context of a post on GitHub. If you feel more competent than FreeCAD devs to decide when or not an issue is a bug in FreeCAD or not, then I would really also expect you to know how to run the supplied command in the terminal. I understand that not everyone knows this, but it doesn't comply with the signals you're sending.

Please remember, that FreeCAD is completely free and an open-source project, developed by people in their free-time. Please show some gratitude towards this, as no one has any obligations to help you at all - why should we use our freetime on this: everyone here could simply had chosen to not care from the beginning.

I'm glad that you got FreeCAD up and running. If you have any future issues, please post them on [forum.freecad.org](forum.freecad.org). @luzpaz, can you close this?

closed 2023-07-31 by luzpaz