FreeCAD/FreeCAD #10010: [Problem] Changing file label and saving the file has no effect

Anthony-Gaudino, 2023-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.2.6
  * Silk 0.1.3
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * Manipulator 1.5.0
  * fasteners 0.4.57
  * lattice2 1.0.0
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * Curves 0.6.12
  * Assembly4 0.50.2
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

When I have a FreeCAD document open, selecting the document, on the data properties panel, there's a field called `Label`, the tooltip says `The name of the document`.
[image]

Changing the value of this field does change the name of the document in FreeCAD but upon saving the file, it overwrites the original file and when closing and opening the file again it remains with the same name.

I though using this field I could set a custom name for the document that only appears inside FreeCAD.

My idea is that that for assemblies I create files named `assy.FCStd` but I can have multiple `assy.FCStd` files open so I could give them a custom name to differentiate them inside FreeCAD, it seems it's not possible to do this.
Also it's not possible to define Descriptions for documents.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-08-30:
We should probably make this field read only

grubertechnic, 2023-08-31:
I second that, changing it makes really no senso at all.

godiard, 2025-05-24:
PR implementing the proposed solution https://github.com/FreeCAD/FreeCAD/pull/21562

commit eb92a4f4 (2025-05-29): App: Make Label property on Document read only (#21562)

Fixes: #10010
closed 2025-05-29 by kadet1090